MarlinFirmware/Marlin #2599: Compile Error when using two extruders

fred2088, 2015-08-19:
mega + ramps 1.4
-----------------------------
temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665: error: unable to find a register to spill in class 'POINTER_REGS'

JamesT42, 2015-08-21:
I see now that this is the same as my issue report
https://github.com/MarlinFirmware/Marlin/issues/2602

thinkyhead, 2015-08-26:
Does this error appear on all versions of Arduino? As I recall this error was caused by a compiler glitch in some versions of Arduino but not others, and can be worked around by rearranging the code a bit.

mkubik, 2015-09-14:
Fails for me as well on a RUMBA board.  This seems to be working on earlier versions of the firmware (pre 1.0.2, can't tell exactly which one.)

Wackerbarth, 2015-09-14:
Gentlemen:

I have NO idea which version of our firmware you are attempting to compile, what configuration file you are using, or what version of the Arduino library you are utilizing.

In particular, EVERYTHING, including our previous RELEASE in this repository is now out-of-date.

We are about to publish a candidate for our next release.  Please try using the "dev" branch in the MarlinDev repository and see if you encounter the same issues.

mkubik, 2015-09-14:
tried that, but it fails with a different error, but it points into the same direction. You can try yourself:
Download latest Dev branch and only change EXTRUDERS to 2 and compile. It'll fail right away. If you switch EXTRUDERS back to 1 it compiles.

Error:

/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/mega -I/usr/share/arduino/libraries/Wire -I/home/mk/sketchbook/libraries/U8glib -I/usr/share/arduino/libraries/LiquidCrystal -I/usr/share/arduino/libraries/SPI /tmp/build6076208065379493663.tmp/blinkm.cpp -o /tmp/build6076208065379493663.tmp/blinkm.cpp.o 
In file included from Configuration_adv.h:601:0,
                 from Configuration.h:860,
                 from Marlin.h:26,
                 from blinkm.cpp:5:
SanityCheck.h:305:8: error: #error TEMP_1_PIN not defined for this board.
       #error TEMP_1_PIN not defined for this board.
        ^
SanityCheck.h:313:8: error: #error HEATER_1_PIN not defined for this board.
       #error HEATER_1_PIN not defined for this board.
        ^
SanityCheck.h:319:8: error: #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
       #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.

Wackerbarth, 2015-09-14:
You are attempting to activate a second extruder on a board configuration that has only one extruder defined.  You need to change your MOTHERBOARD configuration to use a base that has multiple extruders.

mkubik, 2015-09-14:
Well, RUMBA has the capability of running even 3 extruders (see http://reprap.org/mediawiki/images/2/20/Rumba_wiring.svg). Wouldn't it make sense to extend the RUMBA board config in Marlin?

Wackerbarth, 2015-09-14:
Yes, the RUMBA card, and many others, can be configured in a number of different ways. Somewhere, you need to tell the firmware how the wiring is set up for your particular printer.

KiteLab, 2015-09-15:
The errors say: Please define at lease a second temperature sensor.

KiteLab, 2015-09-15:
If you only change extruders to 2 you left the board at BOARD_RAMPS_13_EFB what has only one extruder. try BOARD_RAMPS_13_EEB.

closed 2015-10-14 by AnHardt

jlcdi, 2016-04-04:
"Wackerbarth commented on 15 Sep 2015
Yes, the RUMBA card, and many others, can be configured in a number of different ways. Somewhere, you need to tell the firmware how the wiring is set up for your particular printer."

Hi Wackerbarth, Thank you for this information. But could you be more specific ?
In the MARLIN firmware, the "pins_RUMBA.h" is already prepared for the case where there are many extruders for the RUMBA board. On the line 89 of  "configuration.h", there is the parameter to be entered "#define EXTRUDERS" for the number of extruders connected to the board.
On my RUMBA board, when I enter a number superior to 1, the same error occurs.

In file included from Configuration_adv.h:601:0,
                 from Configuration.h:859,
                 from Marlin.h:26,
                 from blinkm.cpp:5:
SanityCheck.h:305: error: #error TEMP_1_PIN not defined for this board.
     #error TEMP_1_PIN not defined for this board.
      ^
SanityCheck.h:313: error: #error HEATER_1_PIN not defined for this board.
     #error HEATER_1_PIN not defined for this board.
      ^
SanityCheck.h:319: error: #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
     #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
      ^
#error TEMP_1_PIN not defined for this board.

By chance, would you know how to differently tell to the firmware how the board is wired ? In advance, thank you for your help.

emartinez167, 2016-04-04:
We used to have that error with a particular version on the Arduino IDE. Also, have you tried the latest RC5 source code?

I suggest using a different version of Arduino (not 1.6.8 which is the latest as it causes me issues).

Regards,

Ernesto

Blue-Marlin, 2016-04-04:
@jlcdi 
When you increase EXTRUDERS you need the same amount of temperature sensors. Fake sensors (999) are enough.

jlcdi, 2016-04-05:
Hi,

Thanks for your fast answer.

Following your suggestion, I tried with 1.6.0, and 1.6.4 and 1.6.6.

Same result, no matter it is with R5 or with older versions.

1 extruder is OK, 2 or 3 reveals the issue.

I would like to help more, but I can just make the test because the Marlin
Firmware is far beyond my skills.

So, if you need some more tests, feel free to ask.

Best regards.

*Jean-Luc COLONNA d'ISTRIA*
*Director*

*LE MONACO Residences*

*35 Soi Pahonyothin 11, Pahonyothin RoadSamsen Nai, PhayathaiBangkok 10400,
ThailandCell: (66) 8 7972 1828Email: jlcolonna.idts@gmail.com
<jlcolonna.idts@gmail.com>Web: www.le-monaco.com <http://www.le-monaco.com>*

* <http://www.le-monaco.com>SKYPE: bureaubkk*

thinkyhead, 2016-04-05:
I've updated `RCBugFix` so that it gives a proper error message if you forget to set `TEMP_SENSOR_1`, `TEMP_SENSOR_2`, or `TEMP_SENSOR_3`.

jlcdi, 2016-04-06:
@Blue-Marlin   That was not the problem, but thks for this useful reminder.