MarlinFirmware/Marlin #2601: Movement occurring not in G-Code.

ZeroxCorbin, 2015-08-20:
Hello,
First, thank you devs for all your hard work. Really impressive what you have accomplished.

So, I am having a hard time getting complete prints. I have been trying to diagnose what is happening but I am stuck. Hopefully someone can shed light on my mistake.

History: I purchased a Sintron Pursa i3. It came with;
1. Arduino Mega with ATmega2560
2. Ramps 1.4
3. 5x A4988 driver modules
4. LCD 2004 smart display controller with SD card
https://www.3dprintersonlinestore.com/super-smart-prusa-i3

It shipped with Marlin 1.0.0. I upgraded to Marlin 1.0.2. I changed the configuration files to match what Sintron sent me based on Marlin 1.0.0. I would like to share the configuration files but I am not sure how to attach them to this post.

I downloaded the latest Repetier software (1.5.4). I comes with Slic3r 1.2.9. I downloaded a 20mm test cube and tweaked the Slic3r settings to get a good print. I started printing my project and at first things looked good.

I print from the SD card...

After printing a couple of pieces I enabled "Generate Supports" option in Slic3r. I started printing a part and after a few layers the head paused, shook, and then started moving in a random diagonal direction slowly. It dragged across the part. It then moved up a layer, skipping the rest of the layer it was on, and started printing again. I stopped the print.

I started over and got through the print using the same gcode file. The next part I printed was overnight. When I looked in the morning the part had what appeared to be missing or partial layers. There were also a few strands of plastic in random directions out from the part. I tried the print again with the same gcode file. At different locations than before, random movements occurred and layers were skipped.

I decided to try Marlin 1.0.0 that was shipped with the unit. It had similar issues, but it was more with the extrude axis backing up until the filament came out of the head. So i went back to 1.0.2.

I then tried to work with the retraction variables in Slic3r but did not see an improvement. At this point I disabled "Generate Supports" in Slic3r but I would still get the occasional random movement and skipped/partial layers.

I reset Slic3r to defaults and started over thinking I may have messed up a setting. It printed a couple parts correctly but then it started the random movements again. So I read about enabling firmware retraction support in Slic3r. Since I observed the extrude axis backing up farther than it should I thought this would help eliminate a potential Slic3r issue. I enabled the firmware retraction support in Marlin. I was able to get through two prints and things looked good.

I started an overnight print and in the morning the print was 2/3 complete and the X/Y/Z axis where at there positive travel limits and the filament was not in the extrude head anymore. The display said it had completed 97% but the axis where shut off and the heaters were shut off. I went through the gcode and could not find a line that would have commanded the axis to the positions they where at.

Here are some pictures;

[image]

The axis at their limits;
[image]

The filament does not look broken;
[image]

The printer indicates it is still running;
[image]

I have been searching and reading for days. The gcode files Slic3r is producing look good, best I can tell. I have tried many different options in Slic3r and Marlin with not luck.

I may not have given the information needed to determine the issue. If you need anything please let me know and I would be happy to share it. Thank you for your time. Jack..

ZeroxCorbin, 2015-08-21:
I was able to video the issue. At least for the Y axis.

http://42nexus.com/3dprinter/3DPrinter_Runaway.mp4

Could this be an SD card issue?

nitewing76, 2015-08-21:
Since you are using PLA and painter's tape, you need to turn off the heatbed.  

Possibly your PSU is too close to the machine....heat from the CPU is blowing on your controller, the wires laying next to it may be picking up some EMI. 

Since the pic of the RAMPS on the website has heatsinks on the A4988 chips, I will assume yours does also....so you need a fan blowing on these heatsinks (not the one from the PSU).

Remove all excess lubricants...this will attract dirt and debris. Also ensure your lubricant is non-petroleum based.  Even some lithium grease has petroleum products in it (such as the stuff from Lowe's...I can't remember the name, but it has an orange cap). 

Ensure all wires are in twisted pairs...the +/- of limit switches and for the stepper motors, twist the wires that are on the same coil. 

The settings given to you by Sintron for the configuration.h files are only estimations...you need to calibrate to get precises settings. There are many web tutorials on how to do this. 

I suspect your #1 problem is heat, which will be solved with another fan.  If these reccomendations do not work, I would suggest copy+paste your comments and mine to http://forums.reprap.org/list.php?262 . You will get many more responses for your issues there. If you do create a thread there, please post it here so I can follow your progress. :)

ZeroxCorbin, 2015-08-23:
Thank you Nitewing76, I have had a larger fan blowing on the entire control system since the beginning. You can see it in the second comment I posted. The white fan in the upper left.

I have changed to a more modern SD card. I was using an older, 3 years or so, SD card. It was convenient because it flips open and has a USB connection built-in. It seems to have been causing most of the issues. I am using a 30 MBs SD now and it seems to be working much better. I have run about 10 prints and have not had an issue.

In fact, I am closing this issue here and will take your suggestion to post on reprap.org in the future.

Thanks again and have fun printing....

closed 2015-08-23 by ZeroxCorbin

thinkyhead, 2015-08-26:
I can confirm that if you see random movement during a print this indicates a communication / connection / interference problem. There is little that Marlin can do except maybe abort the print as soon as any "bizarre" commands are received. For example, it currently will throw the error __Unknown command: ""__ in some instances. We know this indicates a communication problem, so we could just abort the print job. But I worry that might be too frightening for users, who may just be able to wiggle some wires and continue printing normally.