MarlinFirmware/Marlin #2602: Compiling error for 1.0.x when using more than 1 extruder

JamesT42, 2015-08-20:
Hi!
I'm trying to compile the current 1.0.x version from 20.08.2015 with more than 1 extruder.
Error message:
Arduino: 1.6.4 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:665: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.170 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.170 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out
unable to find a register to spill in class 'POINTER_REGS'

How to reproduce:
Download current version, change the following:
#define MOTHERBOARD BOARD_RAMPS_13_EEB    (or BOARD_RUMBA, havent tested more multi extruder boards)
#define EXTRUDERS 2
And there we go, the error appears!

Wackerbarth, 2015-08-20:
@JamesT42 -- Although our primary effort is now on the 1.1 branch, I did generate a patch on 19Aug that addressed a problem reported by  @radioelf . It is available as PR#2597. Please see if that patch fixes your reported error.

I do not have the hardware to test it myself. So it remains unmerged until someone can verify that it is an improvement and doesn't make the situation worse.

Thanks.

JamesT42, 2015-08-21:
Hi!
The PR doesn't fix my problem.
(I don't know what exactly it fixes, maybe I can test that later after the compilation error is gone)
(You also do not need the hardware, as the compiling error is reproducible without any hardware at all)

thinkyhead, 2015-08-26:
Does this error appear on all versions of Arduino? As I recall this error was caused by a compiler glitch in some versions of Arduino but not others, and can be worked around by rearranging the code a bit.

thinkyhead, 2015-08-31:
In the latest code we avoid the compiler error by throwing an error if you fail to specify a temperature sensor:

    #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.

closed 2015-08-31 by thinkyhead

JamesT42, 2015-09-18:
Why is this closed now? Its still impossible to use more than 1 extruder in 1.0.x

If you can't even fix multiple extruder support, at least tell the users that the 1.0.x branch is dead

Wackerbarth, 2015-09-18:
@JamesT42 -- We cannot declare 1.0.x dead until after we have RELEASED 1.1.x.  However, if you NEED multiple extruders, then you should consider it dead and use the dev version (which is about to be presented as a Release Candidate)

JamesT42, 2015-09-19:
@Wackerbarth How ironic, the dev version can use 3 extruders but it is totally crippled by another bug on my machine.