MarlinFirmware/Marlin #2603: No sd card support on Reprap full graphic display 12864

stefi01, 2015-08-24:
sd card reader detects card, will not open, pronterface detects this and reports a fail and yes it is enabled. latest marlin version

rogerwilco3, 2015-08-25:
the sd card should work ok make sure you use a <2 GB sd card. I have add a old one in my china clone of 256MB  i just recieved there was a 12864 display on uploaded a test file on there and nearly printing out of the box :-)  if i can get the enstops correct. try another card with a lower capacity. i can confirm my sd card works

stefi01, 2015-08-25:
ok thanks for the reply, i will try another 2gb or less card, could i be
using the wrong file name, tried test.g, test.code, init g

thinkyhead, 2015-08-26:
This should be posted to MarlinDev if it's related to the latest code. But I concur with Roger that it's probably not a code issue, but an SD reader issue.

closed 2015-09-25 by AnHardt