MarlinFirmware/Marlin #2604: PID_ADD_EXTRUSION_RATE defined but not used / implemented

docpayce, 2015-08-24:
In Configuration_adv.h PID_ADD_EXTRUSION_RATE is defined and thus Kc and DEFAULT_Kc (Line 21 and on). But the values are not used in the current version of Marlin for any function. I think this might add unnecessary confusion.

Maybe keep this to the dev branch until it is fully tested and implemented? Or at least comment it out and add some flavor text pointing this out.

grob6000, 2015-08-25:
I remember this from a PR I did for PID params per extruder see #1344 #1342. Kc dates back some time, and AFAIK is currently not used in the temp routine. There was some discussion, but no action to remove it. Being relatively new, I didn't want to fiddle more than I needed to either.

I'd recommend posing this in Dev - which will eventually flow through to release.
My preference would also be to either implement fully (as in get the motion planner and PID talking properly - yikes!) or remove fully.

See this for more about Kc and feedforward: http://forums.reprap.org/read.php?147,364712

thinkyhead, 2015-08-26:
Definitely post this as a new issue to MarlinDev. We need more discussion on this, and it relates a little to the JKN Advance algorithm, which also needs to interact with the motion planner in a way that is tricky to implement with the current codebase.

docpayce, 2015-08-26:
Just as a side-note for the MarlinDev - post: With my little background in closed loop control I got (like 10 years ago in university), I tend to believe that a PID control of a known occurence (i.e. extrusion is happening) will always suck balls when compared to directly counteracting that occurence (put in more heat because we are extruding). In other words PID is reacting to the delayed temperature change whereas putting in more heat is actively counteracting the upcoming temperature decrease before it happens. The addtional power needed is very well known and easily calculated. Any difference from exact values will still be corrected by the PID.

Also, when changing PID factors, all three factors should be changed, not only Kp. Just increasing Kp could result in more overshoot and less stability.

I planned to do experiments on this SOME TIME. ;) But I think this could be a nice feature, which might keep hotend temperatures super-stable.

thinkyhead, 2015-08-31:
Okay, so MarlinFirmware/MarlinDev#116 implements this according to the example from the RepRap forum. It probably needs some revision, so please check it out and let me know if you see any problems.

closed 2016-05-04 by jbrazio