MarlinFirmware/Marlin #2606: errors between ug8lib.c and auto leveling enable. ramps 1.4 eeb

batchkrazy, 2015-08-28:
when i go to enable auto leveling feature i get the warnings below. my settings are motherboard ramps eeb/34 2 extruders temp sensors 11 and reprap full graphics smart controller. 

ultralcd.cpp.o: In function `lcd_move_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:734: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:734: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_move_menu_axis':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:705: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:705: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:706: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:706: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:709: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:709: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:710: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:710: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_status_screen':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:266: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:266: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_sdcard_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:959: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:959: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_control_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:745: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:745: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:746: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:746: warning: internal error: out of range error
ultralcd.cpp.o: In function `menu_action_setting_edit_float3':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1039: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_control_temperature_preheat_abs_settings_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:828: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:828: warning: internal error: out of range error
ultralcd.cpp.o: In function `menu_action_setting_edit_int3':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_control_temperature_preheat_pla_settings_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:811: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:811: warning: internal error: out of range error
ultralcd.cpp.o: In function `menu_action_setting_edit_int3':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
ultralcd.cpp.o: In function `lcd_tune_menu':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:421: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:421: warning: internal error: out of range error
ultralcd.cpp.o: In function `menu_action_setting_edit_int3':
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:1038: warning: internal error: out of range error
[... 44 lines cut ...]
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:609: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:609: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:621: warning: internal error: out of range error
C:\Users\cpsim\AppData\Local\Temp\build1913450553732074541.tmp/ultralcd.cpp:621: warning: internal error: out of range error
[... cut ...]
C:\Reprap\arduino-0022\libraries\U8glib\utility/u8g_font.c:1041: warning: internal error: out of range error
C:\Reprap\arduino-0022\libraries\U8glib\utility/u8g_font.c:1041: warning: internal error: out of range error
U8glib\utility\u8g_ll_api.c.o: In function `u8g_init_data':
C:\Reprap\arduino-0022\libraries\U8glib\utility/u8g_ll_api.c:184: warning: internal error: out of range error
C:\Reprap\arduino-0022\libraries\U8glib\utility/u8g_ll_api.c:184: warning: internal error: out of range error
Binary sketch size: 158332 bytes (of a 258048 byte maximum)

Wackerbarth, 2015-08-28:
Specifically, what version of Marlin are you compiling? — In particular, please try applying your configuration to the current “master” from the MarlinDev repository. (And include a reference to the latest commit when you submit a report).
> On Aug 27, 2015, at 9:25 PM, batchkrazy <notifications@github.com> wrote:
> 
> when i go to enable auto leveling feature i get the warnings below. my settings are motherboard ramps eeb/34 2 extruders temp sensors 11 and reprap full graphics smart controller.

closed 2015-08-28 by Wackerbarth

batchkrazy, 2015-08-28:
1.0.2 current release. will download n try the dev build.

paulusjacobus, 2015-08-28:
@batchkrazy When I added my SSD1306 display to the my ramps1.4 board, I had a similar compiler error because the uglib library had an invalid path "utility". The new uglib library refers to "source". I did a find and replace which fixed my issue so you can try this and see if it works.

Wackerbarth, 2015-08-28:
Gentlemen:
We would appreciate your suggestions as to how we can best provide an appropriate “compile environment”.

paulusjacobus, 2015-08-28:
@Wackerbarth the issue here is that the Marlin user needs to import an external library, in this case uglib8 which its author (Oliver Krause) has changed its library structure. Marlin has reference to that structure so a dependency is created which breaks when project Uglib8 releases a new version on Github. My suggestion (not sure if it is possible) would be to remove this dependency in Marlin (so no hard coded paths) but rely on Uglib to provide the library paths or dependencies. Also we can provide feed back to Oliver Krause who created Uglib8 on Github on how to improve his code (if there were such dependencies in his code).

Wackerbarth, 2015-08-28:
I guess that the first question is "How are you attempting to compile the code?"
At the present time, some users are using the Arduino IDE 1.0.5 and others are attempting to use 1.5.x, 1.6.3, or 1.6.6, each of which expect a different file structure.

Even if you upgrade to the latest Arduino IDE, you will eventually run into the issue that "the latest revision of the XXX library" breaks "my" build because my code does not support the recent change.\

paulusjacobus, 2015-08-28:
You're correct that there are dependencies that break with any version of compiler and libraries.
I was more thinking of the less tech savvy people and how to help them. May be we need a troubleshooting guide so they don't have to search through forums.

batchkrazy, 2015-08-28:
maybe i should try using a newer version first. as for i am still using the old 0022 version the arduino ide. gonna try that and get back. i do agree when i got the error with the old ug8lib i downloaded n replaced with the newest version as did i with marlin. never had an issue until i tried to enable the autoleveling feature. and i tried all the different board types for ramps. and came up with the same warnings/errors. which would you recomend. because i have 1.0 as well as the 0022.

paulusjacobus, 2015-08-28:
@batchkrazy version 0022 is ancient and you have highlighted the need for a minimum requirements page (or section) on the main page of Marlin's Github.

batchkrazy, 2015-08-28:
yeah its ancient. that would be a good thing. there are alot of old tutorials out there referencing the old 0022 version.

batchkrazy, 2015-08-28:
ok under 1.0.5 after seeing that i did need to download u8g. it was in fact not an error on u8g but mine not seeing i had updated with the standard avr lib for u8g. so there was no u8glib.h to be found. i updated correctly and it cleared the above errors. so i have updated ide to 1.0.5 now i am getting floating point errors. 
In file included from /Marlin.h:22,
                 from BlinkM.cpp:5:
/Configuration.h:440:9: error: floating constant in preprocessor expression
/Configuration.h:445:13: error: floating constant in preprocessor expression
/Configuration.h:449:9: error: floating constant in preprocessor expression
/Configuration.h:454:13: error: floating constant in preprocessor expression

batchkrazy, 2015-08-28:
and this is what i got out of the current development build. i like the new header file for configuration. it is very informative other than the offsets for the probe. which answered a question i was trying to figure out.
the getting started section looks great. and is something the file has been needing.
here is what i got after changing my probe offsets for ramps eeb and basic settings.
In file included from /Configuration_adv.h:602,
                 from /Configuration.h:859,
                 from /Marlin.h:26,
                 from blinkm.cpp:5:
/SanityCheck.h:181:39: error: floating constant in preprocessor expression
/SanityCheck.h:181:39: error: floating constant in preprocessor expression
/SanityCheck.h:183:42: error: floating constant in preprocessor expression
/SanityCheck.h:183:42: error: floating constant in preprocessor expression
/SanityCheck.h:185:42: error: floating constant in preprocessor expression
/SanityCheck.h:185:42: error: floating constant in preprocessor expression
/SanityCheck.h:187:41: error: floating constant in preprocessor expression
/SanityCheck.h:187:41: error: floating constant in preprocessor expression

thinkyhead, 2015-08-31:
@batchkrazy Yes, that's a limitation of compiler directives. Actually, it's fine to set a #define to whatever you want, but the problem happens if you try to do something like `#if MY_VALUE != 0` and `MY_VALUE` is anything other than an integer. The precompiler (even in 2015) just chokes on it. At least now it throws an error that we can understand!

batchkrazy, 2015-08-31:
yeah c grammar is starting to come back to me. been a long time since i wrote any scripts or code for anything.

thinkyhead, 2015-09-01:
@batchkrazy Welcome back to code-land! I took a long break from coding in the 90's to study philosophy, music, and -um- psychedelics, and then when the Internet started to take off I got a cheap IBM PC knock-off that only ran DOS and QBASIC just so I could get back into practice. Today I'm thrilled that we can download Unity3D (and tools like it) for free and make stuff that blows your mind in a matter of hours. Yay code!

batchkrazy, 2015-09-01:
Yeah started with GW on the commodore back in the day.

batchkrazy, 2015-09-01:
Starting to use catia to design parts again.