MarlinFirmware/Marlin #2608: Auto bed leveling crashes RAMPS.

orrin123, 2015-08-30:
I am very new to 3d printing and this is my first time posting on this form. I apologize if this isn't the right place for this.

I just installed an induction sensor on my printer for auto bed leveling. When I uncomment #define AUTO_BED_LEVELING_FEATURE and upload the firmware to my board the print becomes unresponsive. Nothing shows up on the LCD and my computer wont connect. 

If I comment it out and upload the firmware it immediately begins working again.

My firmware works great otherwise. 
I am using marlin 1.0.2-1 on a RAMPS board with a heated bed. 

Here is my config file: https://drive.google.com/file/d/0B45WRBUZeqjIOU14QURnOFZ0dkU/view
Plz help

paulusjacobus, 2015-08-30:
@orrin123 what printer are you using? Cartesian i.e. reprap or delta ie Rostok?
First thing that catches my eyes is that you haven't defined any temp sensors:
#define TEMP_SENSOR_0 -1
#define TEMP_SENSOR_1 -1
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_BED 0

You have not defined any LCD yet:
//LCD and SD support
//#define ULTRA_LCD  //general LCD support, also 16x2
//#define DOGLCD  //

You haven't defined any servos yet:
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M206 command to correct for switch height offset to actual nozzle height. Store that setting with M500.
//
//#define SERVO_ENDSTOPS {-1, -1, 0} // Servo index for X, Y, Z. Disable with -1
//#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 70,0} // X,Y,Z Axis Extend and Retract angles

You need to uncomment the proper definitions. Hope this helps.

thinkyhead, 2015-08-31:
1.0.2-1 doesn't catch configuration mistakes very well. The next release (1.1.0 coming soon) will prevent you from compiling and give meaningful errors when you forget to set sensors or servos.

orrin123, 2015-08-31:
Thank you,
I was not aware that I had to set any servos because there are non on my machine. I am using a custom reprap.
Thanks again I appreciate the response.

thinkyhead, 2015-09-01:
@orrin123 You don't need to set any servo index if you're using a non-servo probe. You just have to set the sensor pin for it correctly. (You'll only get an error if you tell Marlin you're using a servo but fail to set a servo index and servo pin.)

closed 2015-09-01 by thinkyhead