MarlinFirmware/Marlin #2609: rc servo extruder

giliammc, 2015-08-31:
would it be possible to use a rc servo as an extruder motor

whosawhatsis, 2015-08-31:
Tests have been done, and it works mechanically (see https://plus.google.com/+AnthonyMorris/posts/UeLCSJTePDk and other threads in the same G+ feed), but speed control is an issue. Because continuous-rotation servos do not have closed-loop speed control, it would be necessary to add an encoder to do it properly, the same as using any DC motor for an extruder.

Work has also been done on using a separate arduino to allow a brushed DC motor plus encoder to be controlled with a step/dir signal, emulating a stepper driver. Counting step pulses and running a PID loop to produce steady motion turns out to require a significant portion of an 8-bit microcontroller's processing power, so building an external step/dir driver for this purpose is probably a better option. This also means that no modification to Marlin is necessary.

fmalpartida, 2015-08-31:
Servo with encoder and correct firmware can also replace any axis stepper. The nice thing is you have real loop back feedback.

giliammc, 2015-08-31:
@whosawhatsis the link you gave is dead

thinkyhead, 2015-09-01:
@giliammc The link works for me!

closed 2015-09-01 by thinkyhead