MarlinFirmware/Marlin #2610: Define a custom cancel code for LCD panels

mc71, 2015-09-01:
I'm trying to implement a custom "cancel" code when using an LCD/SD card combo to make sure the printer shuts down correctly.  I can't find where the firmware actions related to "MSG_STOP_PRINT" are linked.

thinkyhead, 2015-09-02:
Are you looking for a code that simulates a click of the LCD panel? Or, are you looking for a code that does a special shutdown different from the usual ending on a G-code file? There is a code, `M112` which calls `kill()` — probably not what you want.

mc71, 2015-09-02:
The functionality would be, when the "stop print" is selected from the lcd menu it calls a safe shut-down sequence, instead of just the typical dead stop.  For me there would be a place to write some g-code sequence which says move up, return to home, retract filamen, turn off motors and heaters, maybe leave the fan on for 3 minutes.

ntoff, 2015-09-04:
That'd be nice, to hit "cancel print" and have the printer pull the nozzle up, shut the heaters off and park it at a certain location. I know of some host softwares that do that, but having the LCD do it would be pretty cool.

thinkyhead, 2015-09-04:
@ntoff At the moment all host software is free to implement that behavior as its own stop procedure, (and we recommend that they do). But we have discussed elsewhere the idea of adding extra moves to the emergency stop in the case of a "thermal runaway" or other bad event. The ball is still in the air about these ideas, so I think it's good to get more input on the subject as we go forward.

ntoff, 2015-09-05:
I'm not talking about the emergency stop though, just the cancel print feature. I might just go add it to my own lcd menu. Cancel print option and a cancel & home option. Yessss. Let the power of the dark side flow through you...

thinkyhead, 2015-09-07:
> just the cancel print feature

I see. The meaning got lost in translation. So @mc71 doesn't need "a code." He needs "some GCode" to run when the cancel option is selected from the LCD. Makes sense to me then.

mc71, 2015-09-07:
Bingo!

As an example of what would be called when you "cancel" a print from the LCD, this is the code we run from our mattercontrol host software.  The user would see a placeholder in configuration.h to insert their own.

G91; Set to Relative position
G1 Z3 E-3; move z axis up
G28  ; home axis
G92 E0; zero the extruded length
G1 F250 E-16; Retract filament from barrel
G92 E0; zero the extruded length again
G90; Set to Absolute position
M104 S0 ; turn off temperature
M140 S0 ; turn off heated bed
M106 S0 ; Turn off fan
M18;

Wackerbarth, 2015-09-07:
Perhaps, rather than doing it in a one-off case, we should create a framework to allow customization of “on-event” actions.
This would be one case.  Beginning and end of prints, Deployment and stowing of the probe, even “Homing”, would be other places that if could be used. And I am sure that others can be identified.

ntoff, 2015-09-08:
Shouldn't you try to keep marlin as small as possible? Wouldn't adding the ability to have 20 lines of gcode in the events add to its compiled size?

Or would you do it that say, if exists cancel.g0, run it, else just heaters off & motors off. ?

Wackerbarth, 2015-09-08:
> On Sep 8, 2015, at 4:46 AM, ntoff <notifications@github.com> wrote:
> Shouldn't you try to keep marlin as small as possible?
> 
That is unclear. We need to decide the minimal HW requirements for newer releases. At some point, we need to quit chasing the idea of adding additional functionality to older hardware. At the present time, my builds are using only 77% of the ROM space. Absolute minimization of code size may not be a good use of our human resources.

It is also unclear how much “extra” code would actually be produced. One “engine” that interprets this micro-coding of operations would handle many different functions. 
> Wouldn't adding the ability to have 20 lines of gcode in the events add to its compiled size?
> 
Actually, even 20 lines of “G-code” can be encoded in a manner that may take less space than the corresponding “C” implementation.
We need to remember that the operations involved are not time-critical. A bit of overhead in the interpretation is unimportant for these operations.
> Or would you do it that say, if exists cancel.g0, run it, else just heaters off & motors off.
> 
You don’t have the additional code for "heaters off & motors off”, per se, instead it is just a micro-sequence that is available for that event.

ghuser2, 2015-09-25:
Here is a similar idea:
https://github.com/MarlinFirmware/MarlinDev/issues/58
http://forums.reprap.org/read.php?2,534549,538584#msg-538584

closed 2016-05-04 by jbrazio