MarlinFirmware/Marlin #2611: If probe is not active auto level fails

rruggiero, 2015-09-01:
Whilst setting up a new printer I found a an issue with the marlin firmware.

When I issue the G29 command and if the the probe endstop is not triggered (not depressed) the G29 command just produces weird behaviour and will end your printer hotends will dance all over the place mid air. After dancing around for a while the hotends will head off toward the top of the Z axis and crash into the top of the printer.  However, if you activate the probe and the endstop is depressed then the G29 command will work properly.

thinkyhead, 2015-09-02:
Are you saying that the probe never reaches the bed while doing `G29`?

What version of Marlin are you using? The current MarlinDev/master has some improvement in leveling compared to 1.0.2.

What kind of printer are you using?

rruggiero, 2015-09-02:
Yes, this is what I found.  I had a video of it which I deleted and I suspect now that the firmware I am using ins not the current one as it was provided to me by Geeetech.

closed 2015-09-02 by rruggiero

thinkyhead, 2015-09-04:
@rruggiero I would suggest trying 1.0.2-1 or 1.1.0 dev because they're easiest to debug, and are "known quantities" — to some extent we know what issues are lingering. As long as they didn't heavily modify it you should find it just works — maybe better, but we will see.