MarlinFirmware/Marlin #2612: Printer halts with latest temperature.cpp

atunguyd, 2015-09-01:
I recently updated my Prussa i3 variant to the latest development Marlin, prior to this I was running a dev version from a few months ago with no issues.

I have THERMAL_RUNAWAY defined for the EXTRUDER but not for the BED.

Initially when I did this upgrade, I rain into a problem when manually moving the extruder head (through the LCD interface) - when I moved the Y axis beyond 50mm, the printer halted with a "Err: MINTEMP BED" on the LCD screen.

I tried a few things but then I though to reset the non volitile memory (Restore Failsafe) and I found that I could now move the head around freely.

I then started a print - after about 3 layers the printer froze and again gave the "Err: MINTEMP BED" message.

Clearly the line of code causing the problem is line 1590 (listed below) but I cant find why this is happening.
if (bed_minttemp_raw GEBED current_temperature_bed_raw) _temp_error(-1, PSTR(MSG_T_MINTEMP), PSTR(MSG_ERR_MINTEMP_BED));

If I diff my old temperature.cpp and this one I cannot yet see a reason for the error (beside the above line being added),

grob6000, 2015-09-01:
Are you aaaaabsolutely sure it's not a physical problem: i.e. an intermittent break in the thermistor wire to the heatbed? The moving of the y-axis on a prusa (bed moves with y-axis) is suspicious...

atunguyd, 2015-09-01:
I can certainly check this, but if I had an intermittent bed connection I would expect by now it would have raised it's head. Nothing was changed in the hardware, only this software update applied.

Nandox7, 2015-09-01:
Do you have the fan starting at that time?
I had mine jumping from 0 to 100 after the first layer (default from some
slicers) and it makes the hotend temp drop fast.
Because it takes some time to bring it back to the expected value it hits
the thermal runaway default timeout.

atunguyd, 2015-09-01:
Nope I dont have a cooling fan in place as I only print ABS. There is a small cooling fan for the extruder cold end and I have defined the pins for this for it to automatically come on when the extruder hits 50 degrees but this happens before the print has even started.

atunguyd, 2015-09-01:
Ok turns out you were right - there was a internal break in the wire going to my bed thermistor - shaking the wire while the bed was heated had the temperature fluctuating - I have fixed the wire and will resume testing.

atunguyd, 2015-09-01:
Huh - dont know how this got reopened - I left my browser on it, must have clicked something accidently

closed 2015-09-01 by atunguyd