MarlinFirmware/Marlin #2613: License violations

Wurstnase, 2015-09-01:
Hi everyone,

do we need a single place for all violations against the license?

Just got an email from someone who told me that Mankati does not send the source code. Only a precompiled hex-file.

Roxy-3DPrintBoard, 2015-09-02:
My thoughts are we do need to maintain a list and keep tract of dates.   I'm thinking we send out maintenance fees and invoices given they aren't living up to their end of the contract.

thinkyhead, 2015-09-02:
We should definitely verify that Mankati is failing to comply with GPL before bothering them. I had a tense exchange with another vendor who it turned out simply wasn't understanding a user request, and I'd hate to repeat that.

thinkyhead, 2015-09-02:
Mankati does looks a little strange!

- Their software is simply a customized Cura with their own branding added.
- No Cura source code is supplied — [unless this is it?](https://github.com/zifengtayu/Cura)
- Their website is exceptionally slow. 45MB download will take >2 hours.
- Comments on YouTube and [reviews in forums](http://www.soliforum.com/post/59481/#p59481) are [mostly negative](http://3dprintboard.com/showthread.php?2983-Mankati-3D-printer&p=47071&viewfull=1#post47071).
- It appears that they don't provide source for their modified Marlin (not yet confirmed that it _is_ Marlin, but is it obvious from the .hex file or startup output?), but according to GPL (as I understand it) they only have to supply it upon request and only to their customers, not necessarily to everyone on their website.

Apparently you can use vanilla Marlin since the Mankati FullScale XT has standard electronics, possibly the same as UM2.

Roxy-3DPrintBoard, 2015-09-02:
> (not yet confirmed that it is Marlin, but is it obvious from the .hex file or startup output?)

I would think the text from Language.h would be very easy to find in the .hex file.   I looked around and  http://forum.mankati.com/files/ was the closest thing I found to them providing firmware.   If there is a .hex file there, I didn't find it with 10 minutes of looking around.

If somebody can provide a link to a .hex file they are offering, I'll check to see if it is generated from a Marlin code base.

Wurstnase, 2015-09-02:
Here is the hex-file, but I will delete this file tomorrow on my drive.
https://drive.google.com/file/d/0B6e9VgA7ItMucTg3QWpwUVhEcWc/view?usp=sharing

Roxy-3DPrintBoard, 2015-09-02:
Thank You!   I have to run to the dentist.   But in a few hours, I'll post the analysis.

epatel99, 2015-09-02:
I was curious so I did a quick check. Here is a part from the running strings on the binary...

<code>
FIRMWARE_NA
E:Marlin V1; Sp
inter/grbl mash
p for gen6 FIRM
ARE_URL:http://
ww.mankati.com/
PROTOCOL_VERSIO
:1.0 MACHINE_TY
E:Mankati EXTRU
ER_COUNT:2 UUID
00000000-0000-0
00-0000-0000000
0000
Mankati O
Mankati read
Bed done.
Bed Heat
Heating don
</code>

Roxy-3DPrintBoard, 2015-09-02:
Here are all the text strings with the associated line of the .hex file.   It looks like Marlin without any Auto Bed Leveling stuff.   

:10013000383420582059205A2045006563686F3AAA  84 X Y Z  E.echo:
:100140000043616E6E6F7420656E74657220737506  .Cannot  enter su
:10015000626469723A20004D3234004D32332025FA  bdir: .M 24.M23 %
:1001600073006175746F25692E67004572726F7236  s.auto%i .g.Error
:100170003A006572726F722077726974696E6720D7  :.error  writing 
:10018000746F2066696C65004E6F74205344207054  to file. Not SD p
:1001900072696E74696E67002F0053442070726933  rinting. /.SD pri
:1001A0006E74696E67206279746520002E00446564  nting by te ...De
:1001B0006C6574696F6E206661696C65642C20469D  letion f ailed, F
:1001C000696C653A200046696C652064656C6574ED  ile: .Fi le delet
:1001D00065643A002E006F70656E206661696C651B  ed:...op en faile
:1001E000642C2046696C653A200057726974696E08  d, File:  .Writin
:1001F0006720746F2066696C653A20002E006F706E  g to fil e: ...op
:10020000656E206661696C65642C2046696C653A90  en faile d, File:
:1002100020002E006F70656E206661696C65642C2D   ...open  failed,
:100220002046696C653A200046696C652073656CF0   File: . File sel
:100230006563746564002053697A653A20004669F5  ected. S ize: .Fi
:100240006C65206F70656E65643A20002E006F70DB  le opene d: ...op
:10025000656E206661696C65642C2046696C653A40  en faile d, File:
:1002600020004E6F772066726573682066696C6542   .Now fr esh file
:100270003A20004E6F7720646F696E672066696C64  : .Now d oing fil
:10028000653A20002220706F730022207061726531  e: ." po s." pare
:100290006E743A2200535542524F5554494E4520F0  nt:".SUB ROUTINE 
:1002A00043414C4C207461726765743A2200747249  CALL tar get:".tr
:1002B00079696E6720746F2063616C6C207375625E  ying to  call sub
:1002C0002D67636F64652066696C65732077697458  -gcode f iles wit
:1002D0006820746F6F206D616E79206C6576656C37  h too ma ny level
:1002E000732E204D4158206C6576656C2069733AF9  s. MAX l evel is:
:1002F0000053442063617264206F6B006F70656E01  .SD card  ok.open
:10030000526F6F74206661696C656400766F6C75FE  Root fai led.volu
:100310006D652E696E6974206661696C6564005351  me.init  failed.S
:100320004420696E6974206661696C006563686F5A  D init f ail.echo
:100330003A0048617264636F646564204465666175  :.Hardco ded Defa
:10034000756C742053657474696E6773204C6F61AB  ult Sett ings Loa
:100350006465640053746F726564207365747469B6  ded.Stor ed setti
:100360006E677320726574726965766564002044F7  ngs retr ieved. D
:10037000002049002020204D3330312050005049CA  . I.   M 301 P.PI
:10038000442073657474696E67733A00205A0020C4  D settin gs:. Z. 
:10039000590020204D323036205800486F6D6520BE  Y.  M206  X.Home 
:1003A0006F666673657420286D6D293A00204500DC  offset ( mm):. E.
:1003B000205A0020580020420020540020204D32B6   Z. X. B . T.  M2
:1003C0003035205300416476616E63656420766148  05 S.Adv anced va
:1003D000726961626C65733A20533D4D696E2066A7  riables:  S=Min f
:1003E0006565647261746520286D6D2F73292C20FA  eedrate  (mm/s), 
:1003F000543D4D696E2074726176656C206665654A  T=Min tr avel fee
:10040000647261746520286D6D2F73292C20423D24  drate (m m/s), B=
:100410006D696E696D756D207365676D656E7420AD  minimum  segment 
:1004200074696D6520286D73292C20583D6D6178A5  time (ms ), X=max
:10043000696D756D205859206A65726B20286D6D45  imum XY  jerk (mm
:100440002F73292C20205A3D6D6178696D756D20C0  /s),  Z= maximum 
:100450005A206A65726B20286D6D2F73292C20201D  Z jerk ( mm/s),  
:10046000453D6D6178696D756D2045206A65726BDB  E=maximu m E jerk
:1004700020286D6D2F73290020540020204D32302C   (mm/s).  T.  M20
:1004800034205300416363656C65726174696F6EFB  4 S.Acce leration
:100490003A20533D616363656C65726174696F6E88  : S=acce leration
:1004A0002C20543D726574726163742061636365CE  , T=retr act acce
:1004B0006C65726174696F6E00204500205A0020DF  leration . E. Z. 
:1004C000590020204D3230312058004D6178696D3F  Y.  M201  X.Maxim
:1004D000756D20416363656C65726174696F6E2030  um Accel eration 
:1004E000286D6D2F7332293A00204500205A0020D4  (mm/s2): . E. Z. 
:1004F000590020204D3230332058004D6178696D0D  Y.  M203  X.Maxim
:10050000756D2066656564726174657320286D6D14  um feedr ates (mm
:100510002F73293A00204500205A0020590020203E  /s):. E.  Z. Y.  
:100520004D39322058005374657073207065722005  M92 X.St eps per 
:10053000756E69743A0053657474696E67732053FD  unit:.Se ttings S
:10054000746F726564006563686F3A004572726F1C  tored.ec ho:.Erro
:10055000723A0053544F505045442E200050726957  r:.STOPP ED. .Pri
:100560006E7465722073746F707065642064756555  nter sto pped due
[... 71 lines cut ...]
:1009E0006973206E6F74204C617374204C696E655E  is not L ast Line
:1009F000204E756D6265722B312C204C6173742012   Number+ 1, Last 
:100A00004C696E653A20004D31313000446F6E659F  Line: .M 110.Done
:100A100020736176696E672066696C652E006F6B66   saving  file..ok
:100A2000004D3239002020506C616E6E6572427547  .M29.  P lannerBu
:100A30006666657242797465733A20002046726575  fferByte s: . Fre
:100A400065204D656D6F72793A20004175672032DF  e Memory : .Aug 2
:100A500035203230313500436F6D70696C65643A12  5 2015.C ompiled:
:100A60002000286E6F6E652C2064656661756C745D   .(none,  default
:100A700020636F6E6669672900207C204175746869   config) . | Auth
:100A80006F723A200041756720323520323031359F  or: .Aug  25 2015
:100A90002031343A34303A353700204C61737420B9   14:40:5 7. Last 
:100AA000557064617465643A2000312E302E300038  Updated:  .1.0.0.
:100AB0004D61726C696E0020536F66747761726568  Marlin.  Software
:100AC00020526573657400205761746368646F67B2   Reset.  Watchdog
:100AD000205265736574002042726F776E206F75C7   Reset.  Brown ou
[... cut ...]
:100AF0006C20526573657400506F77657255700095  l Reset. PowerUp.
:100B00007374617274002200656E717565696E6739  start.". enqueing
:100B10002022002200656E717565696E67202200D3   ".".enq ueing ".
:100B20006563686F3A0020746F6F206C6F6E67208A  echo:. t oo long 
:100B3000657874727573696F6E2070726576656E14  extrusio n preven
:100B40007465640020636F6C6420657874727573DB  ted. col d extrus
:100B5000696F6E2070726576656E746564007C3CAA  ion prev ented.|<
:100B60003E5E2B3D3F2F5B5D3B2C2A225C004530D7  >^+=?/[] ;,*"\.E0

thinkyhead, 2015-09-04:
At this point I have simply asked the basic support question as to which firmware was used, and where the source code can be obtained, to see what their standard response will be. If it is not positive then I will tread lightly. I prefer to inform gently and help them to come over to our way of thinking rather than employ intimidation. That is of course a last resort. For all I know Creatbot/Mankati may only be a very small family outfit, and with a high margin may only care about selling 20 or 30 units a year. Or, they may be a large warehoused company with millions in sales. Either way they should be excited to support openness.

I understand the potential motivation for using Marlin on the one hand (free as in beer) and concealing your "proprietary" customizations (fearing competition) on the other hand. Their modifications are slight and not highly inventive, so I think they over-worry if that is the case. Diplomacy ahoy...

Roxy-3DPrintBoard, 2015-09-04:
Scott, you are always such a nice guy!

leseaw, 2015-09-25:
I become the firmware now :)
all fine for me, will make an fork here for all mankati user.
and the board is  Rumba.

thinkyhead, 2015-09-28:
Thank you @leseaw!

Here is the Mankati / Creatbot fork: https://github.com/leseaw/MarlinDev-Fullscale-XT-Plus

thinkyhead, 2015-09-28:
@leseaw The truly valuable thing will be your `Configuration.h` and `Configuration_adv.h` files configured for those specific machines. Hopefully you can have those posted in your fork soon. Otherwise it will be very hard for users to configure.

leseaw, 2015-09-28:
i must first learn how github work :)
my problem is, if i make an change every time he will share this with marlin ??
next days i will make a new branch with marlin dev and with tmc2100 drivers

thinkyhead, 2016-02-07:
Please post a new issue if this is still a problem with Mankati or anyone else. For now I'm closing this for housecleaning.

closed 2016-02-07 by thinkyhead

ModMike, 2018-09-01:
Sorry to revive such an old thread but I ordered a Creatbot DX today thinking they used Marlin.  Anyone know for sure?

ModMike, 2018-09-01:
The higher models have some kind of color touch screen, kinda threw me off.

InsanityAutomation, 2018-10-02:
In discussion with them on an f430, they told me it is running marlin, so I'd suspect they use it across the board.

ModMike, 2018-10-02:
Wow, you're lucky, they got skittish and refused to answer me.  Spent about 5 full days trying to reverse engineer it.  I don't understand why these companies refuse to work with the Marlin team.  Even if they Patreon Scott, it would be way cheaper than the contortions they go through.

InsanityAutomation, 2018-10-02:
I had asked before creality made headlines for it - all about timing I suppose....

ModMike, 2018-10-03:
My biggest issue right now is that the Creatbot uses a an ADC597 with a K type thermocouple.  I set T0 and T1 to -1 as instructed in the notes.  Tool 0 shows 25 degrees yet I get a MAXTEMP triggered, system stopped.  I tried changing the pins to ADC 10 and 9 but no difference.  Any ideas?

ModMike, 2018-10-03:
Looks like there was an error in the pins files.  Once I set to pin 15, it worked perfectly.  Does that sound right?

ModMike, 2018-10-03:
I assume this is not the proper forum for debugging a Creatbot so I moved it here:  

https://reprap.org/forum/read.php?13,835425,835425#msg-835425

Hope to see a few of you there!

ModMike, 2018-10-03:
@Grogyan YEs it does.  Please go to his link for complete info:

https://reprap.org/forum/read.php?13,835425,835425#msg-835425