MarlinFirmware/Marlin #2615: Displacement during printing

jasane, 2015-09-08:
After doing G29, for bed's leveling, the printing starts and so came to a point where the printer starts to place the object along the heatbed, not at the set up position, it's like it prints each layer at a different position and also ignores the heatbed size.
It feels like the print ignore the G29 for the piece object position.
E.G. When i tried to print a cube it printed it like a maze

thinkyhead, 2015-09-13:
It sounds like something has come loose. Does the layer position vary on both the X and Y axes, or just on one of them?

jasane, 2015-09-14:
Vary in both axes.

AnHardt, 2015-09-26:
Motor current? Printing to fast? Loosing steps?

jasane, 2015-09-26:
Everything on hardware is fine. The belts are fully tight, the current of the motor is adjusted as it should, the axes of the motors are free, the temperature of the motor during work is ok, the maximum velocity is 360 mm per minute and even this still happening. I look up this solves http://wiki.solidoodle.com/fix-axis-shifting and none of them worked for me.

AnHardt, 2015-09-26:
OK.
Without doing a G29 the printer is working fine? How many points do you probe? What G29 do you use 3-point, grid, mesh, manual?

jasane, 2015-09-26:
If i do G29 it gets worse but if i don't it keeps happened but in a less fair amount. I was using grid. But, i'm current not using G29 because i have less problems, but it keep happens

AnHardt, 2015-09-26:
Ok. So it's not a memory overflow. Let's continue in #2645

closed 2015-09-26 by AnHardt