MarlinFirmware/Marlin #2618: G28 runs fine, G29 not at all

GrevanGER, 2015-09-11:
I am stuck with this mistake here and I am at a point where I don't know what to do. 
So I am trying to give as many helpful and detailed information about the situation. 

Typel: 
Hardware: Ramps 1.4 on an Arduino Mega 2560
Host: Repetier-Host V1.5.4

1.: I used "Tom's Guide" to install an inductive sensore along with a heated aluminum bed. 
2.: Wirering seems to be correct, since the sensor does what it is supposed to do when running a G28 
(Along with the M119 of course)
3. When trying to run a a G29 right after a G28, I am still getting "echo:X/Y homing before Z"

I am more than just clue less on what to do about that.

KiteLab, 2015-09-11:
Don't turn your steppers off. 

#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false

GrevanGER, 2015-09-11:
Thanks alot. That did the trick!

closed 2015-09-13 by thinkyhead