MarlinFirmware/Marlin #2619: Head smashed into heatbed with the first line of the objects coordinates

GrevanGER, 2015-09-11:
Greetings. 

I ran into another mysterios obstacle. 
This time, I tried to make my first print. But after the printer ran the G28 and G29 command and gets right to the object at hand, it ignores all boundries and presses the nozzle so hard into the bed, that the printers frame begins to bend. 

This is the gcode of the from the heatvalues until the first line. 
I went trough it step by step, by sending every single line after another to the printer. 
And it is the last one displayed that makes the printer go nuts (or at least the error is attachted to it somehow. 
[quote]M190 S100 ;Uncomment to add your own bed temperature line
M109 S250 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G29 Z0
G1 Z15 F9000 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Drucke...

;Layer count: 336
;LAYER:0
M107
G0 F9000 X100.123 Y76.973 Z0.300[/quote]

thinkyhead, 2015-09-13:
Do you have any offsets applied to your home or nozzle position, such as with `M206` or `M851`? What is the output of `M503`?

GrevanGER, 2015-09-13:
I didn't put any offset up as far as I know. 
But now that I've slept a few nights over it, it might have got something to do with using Cura 15.04 as slicer. That's why the 'G1 Z15 F9000' is commented with " ;move the platform down 15mm ".
But that doesn't explain why the nuzzle crashed so hard into the bed that it bends almost 20°, when it should be going to Z0.300

This is the echo if a M503:

14:04:16.327 : echo:Z-Probe Offset (mm):
14:04:16.327 : echo:M851Z0.00

thinkyhead, 2015-09-14:
I wonder if that `G1 F9000` line is having some weird effect. It shouldn't, of course, but….

Japim, 2015-09-15:
Hi!
Are you sure that you didn't apply any z-offsets? The one applied directly in the Marlin code that defines the distance from z-probe to nozzle end.

What Z-value is shown on the display after the G29 command is executed?
Why you are executing it with G29 Z0?

Line G1 Z15 F9000 says: go to absolute Z 15mm with speed (feed) 9000 - and not to lower it by 15mm.
If it does something else it means that you either didn't correct the z-offsets in the auto-bed leveling, or there are additional z-offsets hidden in slicer setup.

GrevanGER, 2015-09-18:
@Japim: 

Yes, I am sure that I haven't got any Z offsets... 
The quote states my current settings. 

I assume you have just overflown the text. Otherwise you'd have noticed that the command discription for the G1 Z15 is from a makerbot (and these move the platform down instead of the head up).

G29 Z0 is my way of checking if the system took the correct position for Z0. 
So it's just a double check.

GrevanGER, 2015-09-18:
It was the F9000 that caused the problem on my system. 
Everyting is running perfectly now with F set to 1500.

Wackerbarth, 2015-09-18:
Can we close this issue?

GrevanGER, 2015-09-19:
After the third run, the same problem occured again. 

I've double checked the G-Code each time I've moved the SD-Card from my computer to the machine. 
There is no redicioules fast traveling comand left. Yet it's the same as in the beginning.

Wackerbarth, 2015-09-19:
@GrevanGER -- Which version of the code are you using as your base?  I posted Marlin 1.1 RC1 last night. (It had been the master/dev branch for a few weeks.)

GrevanGER, 2015-09-19:
I am not quite sure, since the software was shipped to me by the company that sold me most of the parts. 
But in the Configuration.h is states STRING_VERSION "1.0.2"//"1.0.2"

Wackerbarth, 2015-09-19:
That is an adequate description. We have made numerous changes/improvements since that version was released. We need to determine if those changes affect your issue.

GrevanGER, 2015-09-19:
I see. 
So I guess the best would be to update the whole thing, keeping the old just to keep something to look up the Z axes translation and so on, and go from there.

Wackerbarth, 2015-09-19:
Yes. Obviously you should save the existing stuff in a safe place in case you need to revert to it.
The first step should be to determine the differences between your current Configuration.h (and Configuration_adv.h) and that of the Release. Then update everything to the RC version and apply those differences to that Configuration.

GrevanGER, 2015-09-19:
Okay, I ran into my first problem when trying to verify the changes to the Marlin_RC release.

The symptom: Arduino keeps printing the error message   ```gcode#error "The given LEFT_PROBE_BED_POSITION can't be reached by the Z probe." ``` when reading the equivalent line in the SanityCheck.h

What I did so far:
-Change values in the Configuration.h and Configuration_adv.h to match my hardware
- Installed u8glib again (just to make sure)

My Arduino version is 1.6.5

AlexDoran, 2015-09-21:
Yes, if you read it carefully it is telling you that the Sanity Check has found that the ABL Grid you are trying to use will not physically work based on your Probe - Extruder Offsets and Travel Limits. Try modifying the ABL Grid so that it is at LEAST Bed Size - Probe to Extruder Offset in each direction.

GrevanGER, 2015-09-21:
What probe? 
It is not willing to upload. The error accures whilst trying to upload.
I am not getting the software to upload the new RC software to my machine because of the bug. 

Just in case you ment "Check if your ABL grid matches your travel limit so you can upgrade properly", there is no conflict there.

Travel limits:
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 220
#define Y_MAX_POS 220
#define Z_MAX_POS 160

ABL Grid: 
    #define ABL_PROBE_PT_1_X 15
    #define ABL_PROBE_PT_1_Y 180
    #define ABL_PROBE_PT_2_X 15
    #define ABL_PROBE_PT_2_Y 20
    #define ABL_PROBE_PT_3_X 170
    #define ABL_PROBE_PT_3_Y 20

AlexDoran, 2015-09-21:
Right, so you have a Z Probe for ABL Right?

In the firmware there are offsets to dictate the distance from the Z Probe to the Hotend in each axis:

  // Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25     // Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -29     // Z probe to nozzle Y offset: -front behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.35  // Z probe to nozzle Z offset: -below (always!)

Now, the Sanity Check compares the Grid Probe Position to the Probe Offset to make sure it isn't going to probe outside the area.

Example:

Your first probe point is:
#define ABL_PROBE_PT_1_X 15

Now if your Probe Offset is say 25, like so:
#define X_PROBE_OFFSET_FROM_EXTRUDER -25

Then the firmware knows that the printer is going to attempt to probe 10mm OUTSIDE the print area.

You would have to change the first probing point to at least 25 to make it check out:
#define ABL_PROBE_PT_1_X 25

GrevanGER, 2015-09-21:
There is still no conflict with my travellimits. 

  #define X_PROBE_OFFSET_FROM_EXTRUDER +37    
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -5    
  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0

For X: 170+37 = 207 ; 207< 220
For Y: 20-5 = 15 ; 15>0

So it's all within the limits plus I've got more than just a few milimeters room.

AlexDoran, 2015-09-21:
The problem isn't to do with the Travel Limits.. Try this and see if it compiles:

#define ABL_PROBE_PT_1_X 40
#define ABL_PROBE_PT_1_Y 180
#define ABL_PROBE_PT_2_X 40
#define ABL_PROBE_PT_2_Y 20
#define ABL_PROBE_PT_3_X 170
#define ABL_PROBE_PT_3_Y 20

closed 2015-09-25 by AnHardt