MarlinFirmware/Marlin #2620: Out of filament detect

eboston, 2015-09-12:
I am wanting to add an out of filament function to my Printrbot Simple Metal which is using an older release of Marlin, but hoping I can get some info/help on this.  I have been looking through the code and don't see (didn't expect to) anything to support this ability.  I do see there is a filament change function that would basically do what I want.

What I would like to happen is when the sensor detects the filament has run out, initiate a filament change function (M600).  What I would need to do is have it triggered on a signal to one of the free I/O pins.  Is there any system, at the code level, that would allow me to define a pin and perform an action when it gets triggered?

clefranc, 2015-09-12:
@eboston I think what you're looking for is already in https://github.com/MarlinFirmware/MarlinDev.

eboston, 2015-09-12:
It might be, but the Printrbot is using an older branch and I'm wanting to stay with it for the time being.  I might be able to back port the new changes to my branch.

closed 2015-09-20 by eboston