MarlinFirmware/Marlin #2621: SD Card not working

sarf2k4, 2015-09-13:
Hi, I compared between the stable 1.0.2 and dev 1.1.0, there are no sd card support on dev 1.1.0 but I am able to read files on stable 1.0.2.

I am using:

ramps 1.4
reprap discount 2004 display

I tried copying some of the conditionals for lcd declarations from stable 1.0.2 config.h to dev 1.1.0 and the results was it shows "no sd card" in the menu although the sd card is inserted and no files listed. 

Even I tried enabling some of the individual stuffs on top such as sd support, and ultipanel, it is still the same

thinkyhead, 2015-09-14:
Does it detect if you remove and re-insert the SD card?

But get the latest code and try again. The very latest 1.1.0-dev code forces a check of the SD card at startup, due to changes from MarlinFirmware/MarlinDev#165 and MarlinFirmware/MarlinDev#180. Note that now you _must_ enable `SDSUPPORT` in `Configuration.h`. (This change was made so that `SDSUPPORT` can be disabled with any controller.)

sarf2k4, 2015-09-14:
I will try this again tonight, I noticed last night when I re-insert the sdcard on the lcd panel, marlin somewhat failed to read the files inside and I had to re-insert it again and again until marlin able to read it. I think It's because I'm using an old sd card, 2gb with class4 type an sd card from around 5-7 years ago

sarf2k4, 2015-09-15:
Back to report the results, enabling the sdsupport makes marlin able to read the sd card, but every time the printer turned on from switch, I had to re-insert the sd card and sometimes had to re-insert the card repeatedly until marlin able to read the sdcard

Is this a symptom of old sd card that I need to enable slowsd support in config.h?

AnHardt, 2015-09-26:
<= 2GB, fat-formatted is fine. Defect cards, dirty contacts, broken or long or noisy cableling to the reader are common problems. If the reader can detect the card at all, all recent reading problems have been hardware related.

closed 2015-10-14 by AnHardt

nisraelsen, 2016-04-19:
I had this problem with a Chinese Delta Kossel XL kit with a red Reprap Discount LCD board.  I simply touched a hot soldering iron and a dab of solder to each of the SD card holder contacts on the PCB and . . . voila!  It suddenly started recognizing that I had inserted a card and will now let me print from the card.  Happy day!