MarlinFirmware/Marlin #2622: Megatronics 2.0 Makefile

SquirrelCZE, 2015-09-14:
Hi folks, wanted to try to use Makefile to upload Marlin into my Megatronics 2.0
http://reprap.org/wiki/Megatronics_2.0
compilation works just fine, but upload itself doesn't... and because I am not perfectly skilled in this I would liek to ask for help...
P.S: upload from IDE works, same as I am able to connect to it with octoprint

squirrel@reprap:~/Downloads/Marlin/Marlin$ make upload UPLOAD_PORT=/dev/ttyUSB1 
avrdude -D -p atmega2560 -P /dev/ttyUSB1 -c stk500v2 -b 115200 -U flash:w:applet/Marlin.hex:i
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

Wackerbarth, 2015-09-14:
When you say "upload from IDE works", I assume that you mean the Arduino IDE.

Exactly what message do you get in the log of that upload (It should be very similar to the avrdude message that you quote above)

In particular, I'm interested in the port and baud rate used by the IDE when uploading. (But, please, quote the entire message)

SquirrelCZE, 2015-09-14:
Just
Binary sketch size: 59,816 bytes (of a 258,048 byte maximum)
nothing more outputed by Arduino IDE, sort of ...

anyway, is there soem way to make it output more?
also I found on the wiki page of megatronics mentioned earlier some nice specific setup for megatronics 2.0... I suppose these "setups" are just for arduino IDE?

thinkyhead, 2016-02-07:
Did you ever get this figured out? If not, we can reopen the issue, but for now it's closed for housecleaning.

closed 2016-02-07 by thinkyhead