MarlinFirmware/Marlin #2623: thermal runaway protection

ruggb, 2015-09-15:
I may have mentioned this long ago but I could not find it.
1.0.2-1_de70315 == coreXY bot

As listed below, Marlin.ino will not compile with Arduino 1.6.5
If all 4 are defined or commented out will compile
Is there a reason both bed and extruder MUST be defined for the program to compile?

// Parameters for all extruder heaters
#define THERMAL_RUNAWAY_PROTECTION_PERIOD 40 //**in seconds
#define THERMAL_RUNAWAY_PROTECTION_HYSTERESIS 4 //** in degree Celsius

// If you want to enable this feature for your bed heater,
// uncomment the 2 defines below:

// Parameters for the bed heater
//#define THERMAL_RUNAWAY_PROTECTION_BED_PERIOD 20 //in seconds
//#define THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS 2 // in degree Celsius

If so, why not chg something to implement that with one define to prevent confusion.
Otherwise, it is a bug.

KiteLab, 2015-09-16:
Fixed in 'release'.

ruggb, 2015-09-16:
so why not in dev?

Chaos99, 2015-09-21:
Doesn't look fixed to me.
```
 #ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
      thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, current_temperature_bed, target_temperature_bed, 9, THERMAL_RUNAWAY_PROTECTION_BED_PERIOD, THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS);
    #endif
```
in temperature.h still checks for definition of THERMAL_RUNAWAY_PROTECTION_PERIOD, but then continues to use THERMAL_RUNAWAY_PROTECTION**_BED**_PERIOD causing a *symbol not defined* error if the later is commented out in the config as advised in the help text there.

ruggb, 2015-09-21:
disclaimer: I am not a programmer
Chaos99 - I am confused - u mention the issue in TEMPERATURE.H but I find it in TEMPERATURE.CPP at line 563 in the de70315 dev

 #if TEMP_SENSOR_BED != 0
  
    #ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
      thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, current_temperature_bed, target_temperature_bed, 9, THERMAL_RUNAWAY_PROTECTION_BED_PERIOD, THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS);
    #endif

Shouldn't line 565 be ???
    #ifdef THERMAL_RUNAWAY_PROTECTION_BED_PERIOD && THERMAL_RUNAWAY_PROTECTION_BED_PERIOD > 0

ACTUALLY, that is the fix since it is that way in the release!

If it is that simple - why hasn't it been implemented in dev?

Chaos99, 2015-09-21:
Sorry, temperature.cpp is the right file.

But browsing the file here on the github page with the branch **release** selected still gives me
```
    #ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
```
in line 565. Last commit touching this line is the introduction of this feature from June 2014 https://github.com/MarlinFirmware/Marlin/commit/43c298a7a9674bade1395a75fe54667cdb22fce4

But branch **1.0.x** shows the patch https://github.com/MarlinFirmware/Marlin/commit/1e7b286addc6286d1576ae1a7b68eb2ee2bfb62d applied to line 565.

Seems I misinterpreted the purpose of the branches.  When it says *latest tagged version* the emphasis isn't on *latest*, but on *tagged*. My bad. Should have read further.
Bringing this into the dev branch was not my intention when creating the ticket.
So you might close this on your own discretion.

ruggb, 2015-09-21:
Sorry, my mistake - I am lost in here. I should have put this in MarlinDev - now I see the URL dif.
I will close it and reopen in dev. thx

closed 2015-09-21 by ruggb