MarlinFirmware/Marlin #2628: No SD-Card support for Red RepRapDiscount Full Graphic Smart Controller under 1.1 RC?

GrevanGER, 2015-09-21:
Greetings. 

I've just updated to 1.1 RC thanks to a few helpfull users here. 
But it seems like I am bumping from one problem into the next. 

Riht now I am trying to get my Smart Controller going. 
So in the config.h it says:
The RepRapDiscount FULL GRAPHIC Smart Controller (quadratic white PCB)

And I guess by "white PCB" this line refers this series of Controllers:
http://reprap.org/mediawiki/images/thumb/9/93/RRD_FULL_GRAPHIC_SMART_CONTROLER_INFO.JPG/190px-RRD_FULL_GRAPHIC_SMART_CONTROLER_INFO.JPG

Now my problem is, that I got one of the red ones:
http://forums.reprap.org/addon.php?1,module=embed_images,file_id=38553

The controlls are running fine so far, yet the SD Slot seems not to supported by this 'driver'.

Has anyone got an idea on how to solve this?

epatel, 2015-09-21:
Did you activate SDSUPPORT? Somewhere it seem to have changed so one need to do that explicit

GrevanGER, 2015-09-21:
Yeah. Found it within the massive wall of text right after the Charset settings. 
Uncommenting that line did the trick. Thanks.

closed 2015-09-25 by AnHardt

Reisyukaku, 2015-12-23:
I'm having the exact same problem and none of the SD related defines help.

martinbogo, 2016-08-28:
I'm having the same issue.  I can detect that the SD card is inserted, but cannot actually read a FAT, FAT16, or FAT32 formatted card.  It simply shows "No SD card"

Blue-Marlin, 2016-08-28:
> I'm having the same issue.

So use the provided solution. Activate SD-support.
If even (low level) formatting the card does not help, open a new issue.

martinbogo, 2016-08-28:
I was too terse in my reply -- I am having the same issue after:

= trying both version 1.0.2-1 and the current RC7
= defining SDSUPPORT 
= trying to reduce SPI speed ( half, quarter )
= checking the board to make sure both 3.3v and 5v are present
= defining SPI_SD_INIT_RATE 6
= using a 1, 2, and 4GB SD card formatted FAT, FAT16, and FAT32

thinkyhead, 2016-08-29:
See if low-level formatting the SD card helps. Be sure to back up your data first!

The cardreader library may be "too-quickly" rejecting fragmented SD cards when first inserted. A suggested solution is to try this change in `ultralcd.cpp` around line 2571 (`RCBugFix`):
```diff
      if (sd_status) {
        card.initsd();
+       if (!card.cardOK) card.initsd();
        if (lcd_sd_status != 2) LCD_MESSAGEPGM(MSG_SD_INSERTED);
      }
```

Ilias123povetin, 2017-03-27:
Hello, I changed this line but nothing happend in marlin RC8( M21 works

thinkyhead, 2017-04-02:
@Ilias123povetin How's the `RCBugFix` branch performing at this time?

Ilias123povetin, 2017-04-02:
I will try rcbugfix now and try to formate now)) Sorry for my English I am Russian

thinkyhead, 2017-04-02:
Your English is better than my Russian. If low-level format doesn't address it we can try some other stuff. Usually the problem is caused by an SD card that is too slow to respond on bootup.

Ilias123povetin, 2017-04-02:
I have tried on another marlin version( i choosed some from internet not yours) and the card is working here) But it has a lot of lags and hasnt got  the functions what i need(

Ilias123povetin, 2017-04-02:
And i changed 3 sd cards and all of them are not working(

Ilias123povetin, 2017-04-02:
I think that the trouble is because the reader is learning it very fast and it isn,t in time because when i right m21 comand it see sd card...

thinkyhead, 2017-04-02:
Try this:
``` diff
      if (sd_status) {
        card.initsd();
+       if (!card.cardOK) {
+         safe_delay(1000)
+         card.initsd();
+       }
        if (lcd_sd_status != 2) LCD_MESSAGEPGM(MSG_SD_INSERTED);
      }
```

Ilias123povetin, 2017-04-23:
It didn't helped( Maybe it's because i have a DiscountSmartController not a Fullgrafic

Ilias123povetin, 2017-04-23:
Mb i will send to you the files of the firmware where sd card works?

Bob-the-Kuhn, 2017-04-23:
Did you try RCBugFix?  You can download it [here](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

You'll need to manually move your machine specific items over to the new config files.

You might also look at issue #6249 .  They were having some strange SD card issues.

Ilias123povetin, 2017-05-04:
Noting works(

Ilias123povetin, 2017-05-04:
Nothing works on rc bug fix

SilentZen, 2017-08-02:
I am using a Class 10 card which is detected as inserted but does not auto mount amd says no sd detected.
Using M52 mounts it.
4GB card works fine without M52.
I am on recent bugfix version.
Thanks

Siotma, 2018-01-05:
> I'm having the exact same problem and none of the SD related defines help.
Edit: Seems if i pull it out and insert it numerous time, theres about a 1 in 10 chance that i will get it to register in the menu, but selecting the Print from SD option shows nothing in the root. Tried FAT and FAT32. Not sure if this is just a hardware issue and i got a defect.

I had a very similar issue with a RepRapDiscount Full Graphic Smart Controller and I've just fixed it just adding one delay line as @thinkyhead suggested:

```
if (sd_status) {
+   safe_delay(1000); //FIX FOR MY FULL GRAPHIC SMART CONTROLLER
    card.initsd();
    if (lcd_sd_status != 2) LCD_MESSAGEPGM(MSG_SD_INSERTED);
}
```
To provide more information: I've been using a 4 line RepRapDiscount Smart Controller for 4 years without any problem with different cards, but today I have upgraded my printer with a new one and found the problem: when I turn-on the printer, SD is recognized, but if I try to plug the SD while the printer is turned-on, it does show the "card inserted message" but still "No SD Card" is shown at the Menu.

As I say, adding that line of code at `line 4518` of `ultralcd.cpp` fixes the problem for me.

Bob-the-Kuhn, 2018-01-06:
I've just added that line in bugfix-1.1.x

madmax2, 2018-05-26:
I also have the same issue with my Reprap Discount Full Graphics Smart Controller .

I have tried the code - safe_delay(1000) but still not detecting any external SD card.

I am using marlin Marlin-bugfix-2.0.x version
Board = MKS Sbase 1.3

thinkyhead, 2018-05-26:
@madmax2 — Please file a new issue. Closed old issues don't get the same attention as new open issues.

giancarloandre, 2019-09-01:
My issue is the same and I think it might be a problem with the lcd  since it was printing properly the 2  first weeks then it started with the error “No sd car” even though it detects that the “ sd has been removed” when I remove it and “sd inserted” when I inserted. I tried with differents sd card but no answer seems to come. My next attemp will be to buy a new lcd

gryzlov, 2019-09-28:
> My issue is the same and I think it might be a problem with the lcd since it was printing properly the 2 first weeks then it started with the error “No sd car” even though it detects that the “ sd [...]

Have same issue

giancarloandre, 2019-09-28:
I did change the lcd and still the same problem no possible answer arised. I just kind of resigned to print with the lcd and instead I am printing from the serial port through the arduino and the repetier program

adilrs, 2020-08-26:
// Set USE_SOFTWARE_SPI nonzero to ALWAYS use Software SPI.
#define USE_SOFTWARE_SPI 1   // default 0

// Define software SPI pins so Mega can use unmodified 168/328 shields
#define SOFT_SPI_CS_PIN   53// default 10 // Software SPI chip select pin for the SD
#define SOFT_SPI_MOSI_PIN 51//default 11 // Software SPI Master Out Slave In pin
#define SOFT_SPI_MISO_PIN 50//default 12 // Software SPI Master In Slave Out pin
#define SOFT_SPI_SCK_PIN  52//default 13 // Software SPI Clock pin