MarlinFirmware/Marlin #2629: G29 Auto Bed Compensation Not working

bocamarc, 2015-09-21:
I cannot get G29 command to work.  Nothing happens when I use the G29 command.  I am using a Prusa I3 with Ramps 1.4. The Prusa I3 works properly except for Auto Bed compensation.  I am using Ramps 1.0.2 loaded with Arduino 1.6.5 r5.  The servo works with M401 & M402 commands and homing.  When I use G28 to Home, I get X=39, Y=70, Z=2.3 ( I am using offsets of X=22, Y=0, Z=2.3).  I've tried using G92 to set the home position to X=0, Y=0 and that does not solve the G29 command problem.  How can I get auto bed compensation to work  ?

KiteLab, 2015-09-21:
I suggest before complaining about a defect G29 to repair your G28. Check your endstops.
Or try a M502 M500 first, or disable your EEPROM.
Branch? Configuration? Printer type?

bocamarc, 2015-09-21:
Not complaining. Just looking for help.  My end stops work fine without using auto bed compensation.

GrevanGER, 2015-09-22:
I had my own problem with G28 and G29 a while ago. 

My advice: 
- Make sure that you don't have "Turn inactive steppers off" option NOT active. Set it to flase.
```
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
```
- Run a M119 to check if all your endstops are connected correctly.
Trigger them (hold them down or tape them down) and run another M119 to see if the echo inverts

bocamarc, 2015-09-22:
Disable_X, Disable_Y, and Disable_Z were all set to false previously.

Checking endstops with M119 code show all end stops work properly.  I am currently printing with my Prusa I3 without any problems when auto bed level is turned off.

I have reset the offsets and reloaded the configuration.h to my printer and the following happens - 

---Homing displays (LCD) Z=6.3 (my offset is 6.3), X=39, Y=54  ... 
---G29 Homes the probe and the display shows Z=6.3 (my offset), X=39, Y=54 with message Z end stop hit.
--- The servo/Z end switch works fine  - probe  extends & retracts & switch works properly

I have also tried setting X=0, Y=0 with G92 followed by M500 after I home  Same issue when I home again.

I tested the same configuration.h file with the auto bed level turned off (commented out).  My printer works fine.  When I home it, I get X=0, Y=0, Z=0.  The only difference is that the auto zero is commented out.

Below is the section of my auto zero settings in the configuration.h file.  Please let me know if any other information is needed to understand why I can't get it to work.

Configuration.h Settings ...
//============================= Bed Auto Leveling ===========================

#define ENABLE_AUTO_BED_LEVELING // Delete the comment to enable (remove // at the start of the line)
#define Z_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#ifdef ENABLE_AUTO_BED_LEVELING

// There are 2 different ways to pick the X and Y locations to probe:

//  - "grid" mode
//    Probe every point in a rectangular grid
//    You must specify the rectangle, and the density of sample points
//    This mode is preferred because there are more measurements.
//    It used to be called ACCURATE_BED_LEVELING but "grid" is more descriptive

//  - "3-point" mode
//    Probe 3 arbitrary points on the bed (that aren't colinear)
//    You must specify the X & Y coordinates of all 3 points

  #define AUTO_BED_LEVELING_GRID
  // with AUTO_BED_LEVELING_GRID, the bed is sampled in a
  // AUTO_BED_LEVELING_GRID_POINTSxAUTO_BED_LEVELING_GRID_POINTS grid
  // and least squares solution is calculated
  // Note: this feature occupies 10'206 byte
  #ifdef AUTO_BED_LEVELING_GRID

    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 110
    #define BACK_PROBE_BED_POSITION 110
    #define FRONT_PROBE_BED_POSITION 20

     // set the number of grid points per dimension
     // I wouldn't see a reason to go above 3 (=9 probing points on the bed)
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // not AUTO_BED_LEVELING_GRID
    // with no grid, just probe 3 arbitrary points.  A simple cross-product
    // is used to esimate the plane of the print bed

      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 120
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 120
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER 22
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 16
[... 70 lines cut ...]
//#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 150, 0}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,841}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {250,250,2,22}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {1000,1000,5,1000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          500    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  500   // X, Y, Z and E max acceleration in mm/s^2 for retracts

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
// #define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

bocamarc, 2015-09-22:
Sorry, meant to say toward the end of my comments that auto bed level is commented out not auto zero

bocamarc, 2015-09-24:
Problem solved.  See Issue #2109.  My problem was similar.

I changed the following in the configuration.h file:

const bool X_MAX_ENDSTOP_INVERTING = true; // CHANGED set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = true; // CHANGED set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = true; // CHANGED set to true to invert the logic of the endstop.

The original configuration when I bought the printer, was set to false.  Previously, it had no effect on my use of the printer until I tried to use Auto Bed Level.

When using the G28 G code, my printer does not home at X=0, Y=0.  It now homes at a point in the center of the bed.  The X & Y positions are correct.  Using a G1 X=0 Y=0 the hot end goes to X=0 Y=0.

I had to readjust the Z offset, however, my prints are good.

KiteLab, 2015-09-24:
You could have been faster. See my first message, sentence two.

bocamarc, 2015-09-25:
Your first message was not explained well.  You were not helpful.  I was able to understand the explanation by Thinkyhead.  I have a technical background, however, I am not a developer or  programmer.  I am just getting familiar with the technical side of 3D printing and Marlin.    

Perhaps you need to take a lesson from Thinkyhead on how to provide support & answer a question.  I've looked at some of his other responses and he is very good.  This can be a learning experience for you.  He provides details and explains the problem & solution well.  

You're first response was   defensive & you accused me of complaining. I was not happy with your attitude & did not want to continue to work with you for support.  I looked elsewhere for answers.

I've managed systems engineers & service engineers  at HP, GE, Philips, and Siemens.  If a support person spoke to a customer the way you responded to my request for help, you would have been put on probation.

brainscan, 2015-09-25:
I think people are getting frustrated with new users constantly opening new problems when the solution is already there (and has been for a long time). It wasn't like this a couple of years ago but recently people seem to love blaming marlin when it has more to do with their lack of experience. There are multiple forums to gather info from before opening a new issue.

closed 2015-09-25 by AnHardt

bocamarc, 2015-09-25:
I think Marlin is very good.  I'm finding that there is a good amount of documentation, however,  it is not always consistant/comprehensive.    Since there are many people working on Marlin it is understandable.  I keep finding additional G & M codes not listed and not explained thoroughly.  Many links in the Github issue comments section are dead ends.

In the case of Auto Bed Level, I spent a considerable amount of time on websites and YouTube to try to get it working.  Not one site mentioned anything about endstops.  At first, it was not obvious to Thinkyhead in the previous issue post what was causing the same problem. My configuration.h was provided to me by the company that shipped me the kit.   It had been working fine for the past 2 years.  I had no reason to suspect my endstops were not configured properly.  Guess what, everyone that has received the same generic configuration.h will have trouble adding Auto Bed Level and will either give up or post issue questions. 

 As 3D printing becomes more and more mainstream, the technical abilities of users are less then the early adopters and developers.   It might be worthwhile to have an Auto Bed Level Installation Guide (as well as for other capabilities).  This can be updated as issues arise.  If you have the information correct and in one place, it will elimanate answering the same questions & posts multiple times.  The websites & YouTube videos I reviewed trying to set this up all missed some steps or did not provide comprehensive instructions.  I had to review several sites to get the information needed.  It was both time consuming and frustrating.

Marc Rosenkranz

Wackerbarth, 2015-09-25:
@bocamarc -- I cannot agree more with you. Unfortunately, Marlin is not a commercial product and those working on it are, by nature, very knowledgable on technical issues. Further, on this project, like all others, they seem to be more interested in generating "capability" rather than worrying about "usability" for the less knowledgable. As a result, documentation seems to be the last thing that gets addressed, if it gets addressed at all.

We have also experienced some technical access problems to the documentation files. Someone did some work on documentation in the past but, when they stopped, failed to "return the access keys".

In the last day, or so, we have started to address that problem. I think they are moving items such as you mention into a wiki associated with this repository.

We invite you to add the knowledge that you have gained through your research to the resources available to the next new user.

In some ways, new users are more qualified to decide what, and how, features get documented. Those very close to the code perceive aspects as "obvious" when they are not obvious to those farther removed.

It takes a community to develop a complete product. We invite you to be a part of that community.

bocamarc, 2015-09-25:
Richard,
I would be glad to provide my information to help others.
What's involved in participating  ?  How does one get started.
I will also be doing the following over the next several months:   
   - Upgrading to E3D hot end   

   - Upgrading to Bowden   

   - Upgrading to Dual Extrusion   

   - Building a larger print area/frame   

I anticipate that I will be having a similar experience with each of these upgrades.
I do have an engineering degree, however, it is from ancient times.

Marc

brainscan, 2015-09-25:
If you're thinking of switching to a Bowden for an upgrade(?), I would suggest looking at the flexi drive on the mendelflex. Bowdens will help reduce weight but negatively affect quality, and need lots more trial and error to get things working. The flexi drive keeps best of both worlds, if you actually want an upgrade.

bocamarc, 2015-09-25:
Thank you for the information.  I was not familiar with the Flexi Drive.  It looks good.
I was going to try a direct drive extruder and/or  a modified Wades geared extruder. I wasn't sure which would be better.  I don't want to add any more trial & error or quality issues.  I will be looking more closely at the Flexi Drive.
Marc

Wackerbarth, 2015-09-25:
@bocamarc - "I do have an engineering degree, however, it is from ancient times."
Compared to me, those are likely "modern" times.

The documentation effort IS on the MarlinFirmware/Marlin [wiki](https://github.com/MarlinFirmware/Marlin/wiki) You should be able to find information to help you get started [here](https://github.com/MarlinFirmware/Marlin/wiki/Contributing)

bocamarc, 2015-09-25:
Ok, now I am curious, when did you graduate  ?  I graduated in 1972.

Wackerbarth, 2015-09-25:
BSEE 1965

bocamarc, 2015-09-25:
You got me beat.  I have a BEE with specialty in biomedical.

Wackerbarth, 2015-09-25:
Yeh, biomedical was a new speciality when I was in Grad School. They were major users of "my" computer.