MarlinFirmware/Marlin #2630: G28 Z_RAISE_BEFORE_HOMING Not working

RDickens, 2015-09-21:
Using the latest build and when I issue a G28 command to home all axis the autobed leveller servo tries to swing down but crashes into the edge of my bed and will not trigger. The Z axis is still trying to home and happily crashes into my bed.

Ive tried to set Z_RAISE_BEFORE_HOMING but it does not lift the head.
Using the G29 command the head lifts and lowers correctly allowing the servo to extend and retract.

My home position #define Y_HOME_DIR 1, could this be the issue?

Thanks 
Roy

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB//MOTHERBOARD BOARD_ULTIMAKER
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "This Mendel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// This defines the number of extruders
#define EXTRUDERS 1

//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

[... 693 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H

KiteLab, 2015-09-21:
You ether have not the latest build, or you have copied in an old Configuration.h

RDickens, 2015-09-22:
Sorry rookie i had copied in an old Configuration.h file.
For my own piece of mind ive re-downloaded the latest version 1.0.2 and I still have the same issue 

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.

//#define STRING_VERSION "1.0.2"

#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB//BOARD_ULTIMAKER
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "This Mendel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// This defines the number of extruders
#define EXTRUDERS 1

//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

#define POWER_SUPPLY 1

// Define this to have the electronics keep the power supply off on startup. If you don't know what this is leave it.
// #define PS_DEFAULT_OFF
[... 689 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H

RDickens, 2015-09-24:
After further testing I've discovered that if I home my axis individually then the Z_RAISE_BEFORE_HOMING works fine (G28 X, G28 Y then G28 Z). The problem is when you ask it to home all of the axis at the same time (G28) that it doesn't lift the print head.

Any suggestions?

Thanks
Roy

AnHardt, 2015-09-26:
Especially in the ABL and servo part of the code 1.0.2 is far from good. In the last 9 month since the release of that we worked a lot in that area. Please try or RC. https://github.com/MarlinFirmware/Marlin/tree/RC

RDickens, 2015-09-27:
Put the new 1.1RC on my board and everything seems much better. The G28 moves correctly now however I have now come across another problem.

When I issue a G29 the printer happily levels itself but after it does its last probe I get a message on the LCD panel saying "M105" and it waits for a response on my LCD panel before allowing me to doing anything else.

If I tell it to do a print then the G28 and G29 in the gcode file works fine, it steps through the command sequence perfectly. However when I issue individual commands to it (using pronterface) then it stops after the end of the bed level sequence

Can anyone suggest a solution?

KiteLab, 2015-09-27:
We have seen similar symptoms in #2161.
Do you manually send g-codes while g29 is still working?

RDickens, 2015-09-28:
No I wasn't sending any other codes, it was just a single command at a time.

However, I have now managed to get it working :smile: I have no idea what i had done but the M105 error now was replaced by an "endstop hit: Z" fault after each G29 command. After much searching I found adding a couple of clean up routines in **Marlin_main.cpp** fixed this problem.

```
            apply_rotation_xyz(plan_bed_level_matrix, x_tmp, y_tmp, z_tmp);         //Apply the correction sending the probe offset
            current_position[Z_AXIS] = z_tmp - real_z + current_position[Z_AXIS];   //The difference is added to current position and sent to planner.
            plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
#ifdef Z_PROBE_SLED
            dock_sled(true, -SLED_DOCKING_OFFSET); // correct for over travel.
#endif // Z_PROBE_SLED

//-------New Code----------------
clean_up_after_endstop_move();
endstops_hit_on_purpose();
//--------------------------------------

        }
        break;
#ifndef Z_PROBE_SLED
    case 30: // G30 Single Z Probe
        {
            engage_z_probe(); // Engage Z Servo endstop if available
            st_synchronize();
``` 

Sorry I could not replicate the M105 error but hey I've now got a working printer with an auto bed leveller :thumbsup:

Roy

KiteLab, 2015-09-29:
"endstop hit: Z" is a common problem seen with inductive probes and commented
#define ENDSTOPS_ONLY_FOR_HOMING (Conf..._adv.h)

That combination can not work because the z-min-endsop/probe will trigger every time before the nozzle is close to the bed.

Also permanently triggered when retracted probes (like allen key) can cause this 

Sorry haven't analysed your code jet.

closed 2015-10-14 by AnHardt