MarlinFirmware/Marlin #2632: custom MACHINE

ruggb, 2015-09-22:
in 6effa6f195 = 1.1 RC1
shouldn't this in config....h
line 51  // #define CUSTOM_MENDEL_NAME "This Mendel"
be 
line 51 // #define CUSTOM_MACHINE_NAME "This Machine"

it was changed long ago in dev

also ref in language.h

KiteLab, 2015-09-22:
Would you please be so kind to look into the right branch.

ruggb, 2015-09-22:
I would if I could figure this out

latest commit 6effa6f195
@Wackerbarth Wackerbarth authored 4 days ago 
what does this mean?
That is what I see when I clk on CODE on this page.
and it says BRANCH: RELEASE
help me out!

closed 2015-09-22 by ruggb

Wackerbarth, 2015-09-22:
Yes, it is the RELEASE (1.0.2 was the last RELEASED) RC is only a CANDIDATE for the NEXT release.
You can access any labeled branch from the code page.