MarlinFirmware/Marlin #2634: question re motherboard

ruggb, 2015-09-22:
in 1.1RC1
I used to set my MB = 34 but that has not work for a while now
#define MOTHERBOARD 34
just curious as to why since the MB is defined in Boards.h as 34

Also, why is there this config option in pins.h
// uncomment one of the following lines for RAMPS v1.3 or v1.0, comment both for v1.2 or 1.1
// #define RAMPS_V_1_3
// #define RAMPS_V_1_0

my MB is RAMPS1.4 == RAMPS1.3
since this isn't in config.h, I never changed it, so it is in conflict with my MB define
If it is needed shouldn't it be in config....h?
if not, shouldn't it be gone?

KiteLab, 2015-09-22:
You are referring to 'release' or older.

ruggb, 2015-09-22:
why do u ask?
doesn't 1.1 RC1 define it?
how about 
latest commit 6effa6f195
@Wackerbarth Wackerbarth authored 4 days ago

Wackerbarth, 2015-09-22:
I REPEAT ```6effa6f``` IS NOT 1.1 RC1. It is a derivative of 1.0.2.

closed 2015-09-22 by Wackerbarth

ruggb, 2015-09-22:
OK I think I found the right place
sorry to trouble u
but the last time I clkd on a ...\tree... url it said it couldn't find it.