MarlinFirmware/Marlin #2639: Serial controlled hot end and temp sensor?

delphikwok, 2015-09-23:
Hi,
I noticed that the 1.1RC can support up to 4 extruder. Means it needs 4 pairs of high current wire for the hot ends and 4 pairs low current wire for temp sensors plus fans. And don't forgot hoses for the far end extruder. The increase of wire and hose may increase weight of the moving part. Is it possible to control the hot end and temp sensor and fans by rs232 or UART. Then we just need a pair of power and a pair of TX,RX. 4 wires can control numbers of hotend and temp sersor. Any idea?

Delphi

whosawhatsis, 2015-09-23:
For those conditions, you'd probably want RS485 instead. The Gen3 electronics were designed this way, with an extruder controller (originally mounted on the extruder, later mounted to the side of the bot) connected via UART>RS485 converters to control the heaters and also control the DC extruder motors. This was before the switch to stepper extruders, at which point precise coordination was required, and extruder motor control was brought into the main controller. This left the extruder controller (running on its own atmega168) doing nothing but running a PID loops for the heaters (extruder and platform).  

The machines at that time were single-extruder, so it didn't make a lot of sense, but I've thought a couple of times since then that it might be worth revisiting offloading the PID function for multiple extruders. I've also thought about offloading stepper timing, basically using a network of small, cheap microcontrollers (with enough interrupts for everything) that each do one thing well rather than one big one trying to do everything.

delphikwok, 2015-09-23:
Yup, I understand that marlin is started with single-extruder condition. But for the 1.1RC it support 4 extruders now. So it's better to have serial network for now. I am considering for only the hot ends and temp sensors. so a end to end serial communication is enough. For 2phase stepper extruders requires 4 wire and there is no help to make it RS485 because it still need a pair for power and a pair for +/-. However, if we mount the extruders one by one closely, a multi point network like RS485 does help because it can connect in a daisy chain.  But I prefer CAN network better then RS485 due to RS485 is a master slave protocol. The master require to pull the slave status. I am not sure if the mega2560 still got enough power to do that.

AnHardt, 2015-09-25:
We are making software here.
When we have hardware to test we can return to this.
Closing this until then.

closed 2015-09-25 by AnHardt