MarlinFirmware/Marlin #2640: Missing several comits in 'release'

KiteLab, 2015-09-23:
The 'release' branch compared to the '1.0.x' is missing several bug-fix-commits.
#2266, #2284, #2318, ..., #2584, ...
and some improvements.
Likely an error during recent update of the .md file.

Wackerbarth, 2015-09-23:
Please read the description of the various branch tags.

"Release" is not a branch to which changes are posted. Instead it is a movable tag pointing to the latest tagged release (1.0.2-1 in this case) Since the time that 1.0.2-1 was released, a few additional changes have been made. These are a part of the "1.0.x" branch which IS the latest patched version of that series.  If we were to RELEASE a 1.0.3 version today, it would include these changes and the "Release" branch would be moved to it.

In reality, the next "Release" is likely to be 1.1.0 and the 1.0.x branch will still be around for those who need to access this older base of the firmware.

closed 2015-09-23 by Wackerbarth

KiteLab, 2015-09-23:
No.
1.0.x is your linearised crap from what release used to be.
You reseted 'release' without a good reason and fore the users to download stuff we know is crap.
For the fix points in our history we have the tags - and ziped releases.

Wackerbarth, 2015-09-23:
I don't FORCE you to download any particular branch.  You can download 1.0.x just as easily as you can Release.
 And Release is not sitting directly on top of a tagged release only because it was decided to change the license files and README after the last tagged release. The generated code is unchanged.