MarlinFirmware/Marlin #2643: Configurator is unable to load configuration / can't parse

chaoscrack, 2015-09-23:
Hello

I tried to use the configurator. If I start the index.html I get the information "Configurator is unable to load configuration. Try the upload field". So I choose the configuration.h and get the message "Please upload a boards.h file first". If I do that the next message is "can't parse boards.h"

What should I have to do or what do I wrong?

Thanks for your help

Wackerbarth, 2015-09-23:
The configurator has always been just a "work in progress". It doesn't actually work very well. You will need to perform your configuration manually.

KiteLab, 2015-09-23:
In particular it refers to a not any more existing branch

closed 2015-09-23 by Wackerbarth

thinkyhead, 2015-10-02:
Development will continue on Marlin Configurator as time allows. It is now hosted at https://github.com/thinkyhead/MarlinConfigurator