MarlinFirmware/Marlin #2645: X axis and Y axis Losing coordinates during printing

jasane, 2015-09-23:
X axis and Y axis losing coordinates during printing, the print object get dislocated.
What happens is, it prints half, or an fair amount amount and then goes to some near points and at this new position it keeps printing like nothing happened.

clefranc, 2015-09-23:
@jasane Can it be hardware related. Some loose belt or misadjusted stepper driver?

jasane, 2015-09-23:
Everything on hardware is fine. The belts are fully tight, the current of the motor is adjusted as it should, the axes of the motors are free, the temperature of the motor during work is ok, the maximum velocity is 360 mm per minute and even this still happening. I look up this solves http://wiki.solidoodle.com/fix-axis-shifting and none of them worked for me.

ghuser2, 2015-09-24:
Is the new point on the same layer, or it does it when starting a new layer?

jasane, 2015-09-24:
It happens randomly. It's printing in the current layer, never is at the same layer something is at the bottom layers, other at the top ones, and when it goes to the next layer already starts at the new point.

ghuser2, 2015-09-25:
Sounds like gcode is corrupt.

jasane, 2015-09-25:
GCODE corrupt? What you mean?
I'm using Simplify as slicer maybe it's sending the wrong gcode?
I didn't modified the gcode part on the firmware.

AnHardt, 2015-09-26:
#2615

AnHardt, 2015-09-26:
Let's resume. It's not the Hardware. It's not really related to G29. Your sending correct g-code. Whats left?
...
As this are the typical symptoms of loosing steps, and we do not receive any other reports with this symptoms, my guess is still a hardware problem. 
Do you have a spare driver?

The other likely source of these error is moving, accelerating or jerking to fast.

jasane, 2015-09-26:
My boss just told me he is trying put 2 more pull ups because he's now strongly believing that maybe has an interference at the hot end. I'll send any further information on monday.
Thank you everybody.

AnHardt, 2015-09-26:
Do you set up this printer for the first time, or did it work before?

Did you check the voltage of the power supply under full load - when extruder and bed are heating at full power?

dteeter, 2015-09-27:
I've had this exact problem before. For me it happened because the stepper current was turned down too low and the motors were skipping steps. Other reasons for skipping steps can be: extruder getting hung up on warped edges, mechanical binding during a movement, overheating of drivers, slippage of belt sprockets on the motor shaft, etc.

closed 2015-10-14 by AnHardt