MarlinFirmware/Marlin #2646: 1.1 RC - Offset?

GrevanGER, 2015-09-23:
Slicer: Cura 15.04.2

So, after solving a series of problems that had to do with my lagg of experience, I ran into another "problem" that might just have got something to do with me being a beginner with marlin. 

So the problem is, that I don't seem to be able to influence, where the print is placed on the bed. 
It is always being placed "touching" Xmin. 

What I have tried so far: 

- Placing the objects at the "positive" end of X within Cura
- Toogeling the Cura internal option to define if my printer is placing X0.0Y0.0Z0.0 at the center of the bed or not.
- Trying to define an offset via "#define X_MIN_POS 40" 

Would be nice if someone could point me in the right direction to solve this.

GrevanGER, 2015-09-24:
After several tests today, I come to the conclusion, that this is a problem with Cura 15.04.2

- I've sliced several different objects, all with the same outcome: The print starts at Xmin and center Y. 
- Then I gave RepetierHost a try. Result: Even with Hystersis 1 and Period 40, the printer gives me a "Heating failed" error. 
- After reading myself into the expert settings of the newest Cura version (15.06.03) I gave that a try. Result: All prints have been place along Ymin, completely ignoring the heatup phase and G28 and G29
BUT
When I use my printers smart controller to home and heat up before loading the gcode for an object from the SD card, I can print just fine. 

It still gives me a headache since I'd like to be able to do it the regular, easy way by just loading everything from SD and having the Gcode homing and heating my macine AND starting to print the objects where I put them on the bed in the slicer.

GrevanGER, 2015-09-25:
Now I am stuck with the question: 
Which Slicer did the dev them use when coming up with this version and which settings (aside from accileration and co) did you use to make it running? 
It's pretty frustrating to have everything running but not being able to transfer my objects to the printer properly.

clefranc, 2015-09-25:
@GrevanGER The printer prints where the G-code commands commands it. The slicer slices your object where you ask it to slice it. The printer can only add basic offsets or bed levelling correction to your prints, not transform or translate to a new set of coordinates.

It's looks like a slicer issue to me.

GrevanGER, 2015-09-25:
So which slicer would you recommand to use?

clefranc, 2015-09-25:
I'm a lazy guy: Repetier-Host with Slic3r.

closed 2015-10-14 by AnHardt