MarlinFirmware/Marlin #2648: info needed

ruggb, 2015-09-24:
can someone tell me how to find this - I have been searching for an hour
hardware/marlin/avr/libraries/marlin/examples/MarlinFirmware/FirmwareCustomization.h

clefranc, 2015-09-24:
@ruggb I presume you're trying to compile the "new system" of MarlinDev firmware?

GrevanGER, 2015-09-24:
https://github.com/MarlinFirmware/MarlinDev/tree/master/examples/MarlinFirmware

BuzzBumbleBee, 2015-09-24:
@GrevanGER yeah its a little confusing afaik marlin dev needs you to add it to the lib folder combined with https://github.com/MarlinFirmware/MarlinDev/tree/IDE_1.6.5/

ruggb, 2015-09-24:
thx, somehow I finally stumbled across it. I wanted the info as ref on
https://github.com/MarlinFirmware/MarlinDev/
but this just keeps getting more confusing.
I hope someone can author a good tutorial for us dummies.

closed 2015-09-25 by AnHardt