MarlinFirmware/Marlin #2649: Feature Request: Tool Offset (with dual X-carrige)

pbrgle1, 2015-09-24:
With the dual x-carrige, the tools can have a different Z since they move independently (not in duplication mode). I would like to request a feature that enables Z (and maybe also Y) offset right after tool change.

Wackerbarth, 2015-09-24:
This sounds like it is something that is beyond the scope of a Marlin based printer. Can you be more explicit in describing the feature and how it would work?

pbrgle1, 2015-09-24:
well the dual x carrige mode is already implemented in marlin under configuration_adv.h (#define DUAL_X_CARRIAGE), and so is the x offset of the parked extruders. I would also like to see some feature that specifies the z offset of the nozzles. For eg. Tool 0 is on Z 0.00 when homed and Tool 1 has a -0.4 offset. So everytime the tool changes, there would be a Z lift or lowering according to the offset.

Wackerbarth, 2015-09-24:
OK, I think I understand your request. You want a per-tool "height setting" when the carriages are operating as separate tools. Thus switching from one tool to another would not only change the X-stepper motor involved, but would change the Z-position so that the resulting nozzle is at the same height in the object space.

pbrgle1, 2015-09-24:
Exactly. So if Tool 1 is 0.5mm lower - before switching to Tool 2, perform a 0.5mm Z lift and vice versa.

Wackerbarth, 2015-09-24:
I would approach it a little differently, but with the same effect.

pbrgle1, 2015-09-24:
Is there any chance for this to be implemented?

pbrgle1, 2015-09-24:
i just found this. https://ultimaker.com/en/community/view/16095-marlin-m218-extruder-offsetdocumented-as-only-x-y-but-the-code-says-also-z?page=1&sort=#reply-119605.

is there anything to it?

KiteLab, 2015-09-24:
https://github.com/MarlinFirmware/MarlinDev/blob/master/Marlin_main.cpp#L4078
Try it. At this place we have the same code.

pbrgle1, 2015-09-24:
I set the Z offset to -0.5 and M114 Reports Z0,00 for tool 0 and Z0,50 for tool 1 even though the Z didnt move....what could be wrong?

clefranc, 2015-09-25:
@pbrgle1 Which command did you use to set the Z offset?

pbrgle1, 2015-09-25:
M218 T1 Z-0.5

clefranc, 2015-09-25:
T1 F

pbrgle1, 2015-09-25:
F?

clefranc, 2015-09-25:
I don't know if you're using a host or direct command. The F in T command force the move when changing tool.

pbrgle1, 2015-09-25:
im using pronterface atm.

clefranc, 2015-09-25:
Still not moving?

pbrgle1, 2015-09-25:
hmm, no. the M114 does report the z change when i switch tools but the printer doesn't move Z...

clefranc, 2015-09-25:
If you set an X and Y offset for T1, do it move?

Sorry, I've not experience with dual carriage, looking at the code seems to indicate it's implemented.

pbrgle1, 2015-09-25:
Nope, there are no offsets.

clefranc, 2015-09-25:
Can you paste your Configuration.h and Configuration_adv.h here (or create a branch in your repo with your config)?

clefranc, 2015-09-25:
Configuration_adv.h?

clefranc, 2015-09-25:
Deleted?

pbrgle1, 2015-09-25:
hey. i dont know what happened... github thought i was  abot

clefranc, 2015-09-25:
Hey! Welcome back.

I thought you solved your issue and kept the solution for yourself. :)

@KiteLab and I wanted you to try a G0 X0 Y0 Z0 after a tool change.

AnHardt, 2015-10-14:
No reaction.

closed 2015-10-14 by AnHardt

nicksears, 2016-07-10:
Not sure if this is completely relevant, but I think some people will find this when looking for my similar issue. I don't have a dual carriage, but I still need to be able to offset in Z.

To force this ability, you just have to comment out the conditional in three places in marlin_main.cpp:

//#if ENABLED(DUAL_X_CARRIAGE)
and each corresponding 
//#endif

also posted on https://ultimaker.com/en/community/16095-marlin-m218-extruder-offsetdocumented-as-only-x-y-but-the-code-says-also-z

thinkyhead, 2016-07-11:
@nicksears What kind of extruder(s) are you using? Upcoming support for switching extruders will be adding a generalized Z offset as `HOTEND_OFFSET_Z`.

nicksears, 2016-07-11:
I'm printing paste and a thermoplastic with HYREL's EMO-25 emulsifiable extruders. 
I'll post details when I publish :) but the paste extrusion part can be found [here].(http://onlinelibrary.wiley.com/doi/10.1002/marc.201600236/abstract)
Since I replace the entire extruder so often, zeroing the heads perfectly every time isn't feasible
I'd rather get it close and measure the offset perfectly with a probe and webcam.
I don't see a reason to implement a separate z offset methodology when it's already built in?
Why not just have a "#define ALLOW_HOTEND_OFFSET_Z" 
and change the if statements to "#if ENABLED(DUAL_X_CARRIAGE || ALLOW_HOTEND_OFFSET_Z)"?

thinkyhead, 2016-07-12:
@nicksears Heck, we'll probably just make `HOTEND_OFFSET_Z` standard in all cases, but leave it zeroed out by default.

nicksears, 2016-08-09:
Just a note about this. I don't know if the HOTEND_OFFSET_Z was supposed to make things better/more universal, but it broke what I was doing before. Now in RC7 I may be able to set a hotend offset, but it doesn't work as expected. It never seems to apply it. 

I've made it work by defining as a switching_extruder since dual_x_carriage and switching_extruder seem to have a unique way of applying the offset during a tool change in marlin_main.
I've had to tweak a lot of other things since I'm not actually using a switching extruder, (especially making the stepper behavior like normal in stepper_indirection.h) but it seems to work again. Does this sound right?

thinkyhead, 2016-08-09:
> but it broke what I was doing before
> but it doesn't work as expected

These are awfully vague statements. What is the specific bug?

nicksears, 2016-08-09:
Before I was able to just comment out the if statements in M218 in marlin main to allow z_offset with M218. I'm not sure it was working completely correctly, though. The height readout would show the current position with the offset included, and I thought that was supposed go be internal.

In RC7 it showed the correct height, but never actually applied any offset (just auto bed leveling compensation). Both SWITCHING_EXTRUDER and DUAL_X_CARRIAGE have sections in the tool change method that shift it to the new location (including z) but since those are the only cases that allow for an offset z, it seems that it wont apply the z offset unless you have one of those selected. Also, the section after these for AUTO_BED_LEVELING_FEATURE no longer applied the offset in z? not sure if that's relevant.

So I selected SWITCHING_EXTRUDER, commented any corresponding servo actions, commented the first part of the if statements in stepper_indirection.h that limits it to one motor that inverts direction, and lots of other small changes that caused errors. 

I think it works now, but avoiding collisions is difficult. In practice it should never be a problem because my extruders are set wide apart, but until I have the logic figured out, I'm not confident.