MarlinFirmware/Marlin #2651: Where is issue #2649?

clefranc, 2015-09-25:
@pbrgle1 Where is your issue #2649?

KiteLab, 2015-09-25:
Server error?

Changing the offset is one thing, actually moving the head is an other.
I would be interested how the head is moving in the next move - for example a G0 Z0.

clefranc, 2015-09-25:
@pbrgle1 closed its GitHub account, issue gone with it. Is this common?

t-est-user, 2015-09-25:
#2652
test

KiteLab, 2015-09-25:
ghost was @t-est-user.
#2652

KiteLab, 2015-09-25:
Are you sure you want to do this?
    This is extremely important.
    
    We will immediately delete all of your repositories (1), along with all of your forks, wikis, issues, pull requests, and GitHub Pages sites.
    
    You will no longer be billed, and your username will be available to anyone on GitHub.
    
    For more help, read our article "Deleting your user account".https://help.github.com/articles/deleting-your-user-account/

At lest you are warned that your issues will disappear
I really don't know what the difference between @pbrgle1 and @t-est-user was.

clefranc, 2015-09-25:
@pbrgle1 forked this repo.

KiteLab, 2015-09-25:
@clefranc 
Good argument. But i'll not test this. Let's close this.

closed 2015-09-25 by clefranc