MarlinFirmware/Marlin #2653: Dual Z Axis Stepper, Random Switching Direction

AlexDoran, 2015-09-25:


KiteLab, 2015-09-26:
Randomness is always hard to analyse. Let's change the likelihoods.
Is that reversing correlated to high speeds on the z-axis?
Play with z-speed and z-acceleration, z-jerk.
Is the reversing motor running smooth? Any mechanical resistances?

AlexDoran, 2015-09-27:
No fast movement or acceleration, mechanically everything is sound. However the print has a 450 x 450mm print area, and i recently installed a 450 x 450mm 240v AC heated bed.. could this be some kind of interference?

Thanks

Alex

AnHardt, 2015-10-09:
Noise on the endstops line is a common problem. I do not see how this can reverse a motors direction, but i'm not that familiar with the dual-Z code.
Tho be sure you could look if `ENDSTOPS_ONLY_FOR_HOMING` is defined. If not - please uncomment.

closed 2015-11-24 by AnHardt