MarlinFirmware/Marlin #2656: M218 for Z offset not working with dual X carriage

pbrgle1, 2015-09-25:
This is realated to the previous issue that i posted. In marlin_main.cpp i founf that someone probably already tried to implement this feature but it wasn't documented and probably not fully implemented.

/**
 * M218 - set hotend offset (in mm), T<extruder_number> X<offset_on_X> Y<offset_on_Y>
 */
inline void gcode_M218() {
  if (setTargetedHotend(218)) return;
  if (code_seen('X')) extruder_offset[X_AXIS][target_extruder] = code_value();
  if (code_seen('Y')) extruder_offset[Y_AXIS][target_extruder] = code_value();
#if ENABLED(DUAL_X_CARRIAGE)
  if (code_seen('Z')) extruder_offset[Z_AXIS][target_extruder] = code_value();
#endif
  SERIAL_ECHO_START;
  SERIAL_ECHOPGM(MSG_HOTEND_OFFSET);
  for (int e = 0; e < EXTRUDERS; e++) {
    SERIAL_CHAR(' ');
    SERIAL_ECHO(extruder_offset[X_AXIS][e]);
    SERIAL_CHAR(',');
    SERIAL_ECHO(extruder_offset[Y_AXIS][e]);
#if ENABLED(DUAL_X_CARRIAGE)
    SERIAL_CHAR(',');
    SERIAL_ECHO(extruder_offset[Z_AXIS][e]);
#endif
  }
  SERIAL_EOL;
}

I tried to use M218 because my nozzles are 0.5mm apart in my dual x carriage system. The M114 does report that the height has been changed, but in reality nothing happens. Its probably not very difficult to fully implement also the Z offset and it would ber a huge + when dealing with dual x carriages. Right now i am waiting for a post-processing script from a friend because i can't code (yet) but i would like to see this fixed.
 
Sorry if its a repost, just wanted to state the issue more clearly.

AnHardt, 2015-09-26:
Where will a G0 Z0 end after your M218? Changing the offset and moving to there is not the same.

AnHardt, 2015-10-14:
No reaction

closed 2015-10-14 by AnHardt