MarlinFirmware/Marlin #2660: Axis attempts to go into the -home position after homing while bed leveling

Surgikill, 2015-09-27:
I have a dual y axis setup on my printer using a Rambo board. After the axes home it attempts to bed level. In doing this it tries to send the print head past the minimum endstops on the x and y axes, while the z axis stays stationary.

AnHardt, 2015-09-27:
Dual-y  that's pretty rare - so likely tested nod very well.
Please tell us the version/branch you are using. Your Configuration.h and in this case Configuration_adv.h will help us to reproduce this error (We prefer gist or dropbox links where we can directly download). The exact sequence of g-codes you are sending from the last boot to the error may be relevant. Tell us about your machine. What endstops - probes are connected?

AnHardt, 2015-10-14:
No reaction

closed 2015-10-14 by AnHardt