MarlinFirmware/Marlin #2662: add help command

giliammc, 2015-09-28:
i have a printrboard and my current version of marlin only takes up 60% of the available space so i was wondering how i could add a help command that would display all of the g codes for marlin

AnHardt, 2015-10-05:
As you need a host to input a g-code into Marlin it's much more convenient to use a existing website.
I suggest
http://reprap.org/wiki/G-code
and
http://www.marlinfirmware.org/index.php/G-Code_in_Marlin https://github.com/MarlinFirmware/Marlin/wiki/G-Code-in-Marlin
when ready one day. Do you want to help here?

giliammc, 2015-10-06:
what do you mean "help here?"

AnHardt, 2015-10-06:
Writing the wiki.
All of the wikis are incomplete. :-(
The first and the third are writable for all.
All the info about the g-codes is in 'Marlin_main.cpp' There is a big info block at the top of the file and absolute every detail, of the implementation of the g-codes, in the code below. :-)

Even if you don't trust your 'English' - for the average user that will be better readable than the code itself.

closed 2015-10-14 by AnHardt