MarlinFirmware/Marlin #2663: Support for STM32 Chip : News

Drakelive, 2015-09-28:
It is some weeks I'm trying various Arduino Hardware CORE for STM32 chip to figure out what the easiest and the most comprehensive.
What seems to be the most comprehensive it is developed by the community of Roger Clark: https://github.com/rogerclarkmelbourne/Arduino_STM32
My tests were performed using a CZ Mini Board, based on STM32F103: www.stm32duino.com/viewtopic.php?f=28&t=295
The Blink Example works perfectly, even changing Pins and Ports of the board and the community (www.stm32duino.com) has simple sample projects.

We can install the CORE on the latest version of the Arduino IDE and provides support for board based on ARM chip like a F1, F3, F4, microduino,
maple, maple mini, maple Ret6 and ST NucleoF103RB but this requires enabling the internal clock following a easy procedure fully described.

I know this is a topic already discussed but I think this CORE can be extremely valuable to start using it in Marlin to have a version of ARM (STM32)
I would propose to start with a general release based on F1 chips that are fairly inexpensive and can be found easily in the Chinese market.

thinkyhead, 2016-02-08:
There has been some discussion about supporting newer architectures. In general, most favor starting with a new code-base, and in fact some work is already being done in this area by @ErikZalm and possibly others. I suggest you join into the discussions on this topic, or start a new thread in the MarlinDev tree. Cutting-edge development should be discussed there instead of here, where we deal mainly with fixing bugs in releases.

closed 2016-02-08 by thinkyhead

egorse, 2016-03-04:
@thinkyhead Please check hex of GTM32 - http://www.geeetech.com/forum/viewtopic.php?f=20&t=17279 - IMO that is Marlin running on STM32 but there is no source code published

Blue-Marlin, 2016-03-04:
How about https://github.com/search?utf8=%E2%9C%93&q=STM+Marlin&type=Repositories&ref=searchresults

CONSULitAS, 2016-03-04:
@Blue-Marlin 
It says `not working for now.`.

@egorse 
That IS Marlin. Perhaps you can ask them for the source code.

Because Marlin is under the GPL License, this has to be GPL, too! If they don't want to publish their source, you should remind them of the GPL and that it would be a license violation not to give the derived code.

egorse, 2016-03-05:
@CONSULitAS made #3081