MarlinFirmware/Marlin #2665: Autolevel adjust incorreclty

Sungod3000, 2015-09-30:
Hi,

I test this version and while trying to find the best auto level routine. The g30 and g29 work reliably and as expected. When I visualize the result of g29 I get a repeatable pattern. 

I printed some smaller items over the last days and it worked well however im noticing now that I print bigger item that the plane is quite tilted. I checked the marlin wiki to find out more, but it only lists the g29 parameters.

I didn't find a way to check what leveling compensation matrix is applied or is there a parameter that applies the matrix that is not included in the default g29?

AnHardt, 2015-10-05:
" this version" is not enough information. MarlinFirmware/Marlin has today 4 branches and a couple of .zips you could have downloaded.
If you are on one of the newer branches a "G29 V4" may tell you more about the matrix and the height map. Remember not to G28 after a G29 - else the matrix is reset. If the height-map appears not only tilted but saddled ABL can not do much against that. 

A link to your 'Configuration.h' and a printout of "G29 V4", or at least "G29" will help us to analyse your case.

Sungod3000, 2015-10-05:
my bad I thought each branch has its own issues section. im using 1.1 RC1
-g29 is only tilted http://www.maui-3d.com/cgi-bin/plotG29?plotG29ID=5370
-no g28 after g29
-

SENDING:G29 V4
G29 Auto Bed Leveling
Bed X: 80.000 Y: -84.000 Z: 1.719
Bed X: 39.000 Y: -84.000 Z: 1.200
Bed X: -2.000 Y: -84.000 Z: 0.981
Bed X: -43.000 Y: -84.000 Z: 0.725
Bed X: -84.000 Y: -84.000 Z: 0.450
Bed X: -84.000 Y: -43.000 Z: 0.438
Bed X: -43.000 Y: -43.000 Z: 0.787
Bed X: -2.000 Y: -43.000 Z: 1.012
Bed X: 39.000 Y: -43.000 Z: 1.275
Bed X: 80.000 Y: -43.000 Z: 1.556
Bed X: 121.000 Y: -2.000 Z: 1.700
Bed X: 80.000 Y: -2.000 Z: 1.175
Bed X: 39.000 Y: -2.000 Z: 1.000
Bed X: -2.000 Y: -2.000 Z: 0.794
Bed X: -43.000 Y: -2.000 Z: 0.575
Bed X: -84.000 Y: -2.000 Z: 0.325
Bed X: -84.000 Y: 39.000 Z: 0.431
Bed X: -43.000 Y: 39.000 Z: 0.644
Bed X: -2.000 Y: 39.000 Z: 0.912
Bed X: 39.000 Y: 39.000 Z: 1.056
Bed X: 80.000 Y: 39.000 Z: 1.212
Bed X: 80.000 Y: 80.000 Z: 1.106
Bed X: 39.000 Y: 80.000 Z: 0.900
Bed X: -2.000 Y: 80.000 Z: 0.762
Bed X: -43.000 Y: 80.000 Z: 0.575
Bed X: -84.000 Y: 80.000 Z: 0.337
Bed X: -2.000 Y: 121.000 Z: 0.969
-0.87 -0.64 -0.42 -0.13 0.04 0.31 1.08
-0.91 -0.63 -0.36 -0.10 0.12 0.64 1.03
-0.99 -0.64 -0.29 -0.07 0.19 0.48 0.76
-1.01 -0.76 -0.51 -0.29 -0.08 0.09 0.62
-0.86 -0.65 -0.44 -0.17 -0.02 0.13 0.29
-0.79 -0.74 -0.51 -0.32 -0.18 0.03 0.23
-1.05 -0.84 -0.57 -0.11 -0.34 -0.19 0.08

KiteLab, 2015-10-05:
Ether this is not the printout of a "G29 V4" from 1.1 RC1, or you edited almost all information away.
It looks more like from a much older branch.

KiteLab, 2015-10-05:
Ah - sorry. You have a delta? There is no matrix for deltas. ABL works totally different there.

Sungod3000, 2015-10-05:
Yes delta.

Roxy-3DPrintBoard, 2015-10-05:
SunGod3000 Do you have your bed mounted with 3 Screws around the perimeter?   If so, I may have a piece of code that will help you manually level your bed so you don't need the Auto Bed Leveling.

Sungod3000, 2015-10-05:
I have levelled it manually and it works (never the less Im interested in your code). So ABL is not necessary but its really convenient.

Roxy-3DPrintBoard, 2015-10-05:
OK, Understood.   I'm bringing my Delta printer up but it isn't fully functional yet.  So I can't actually dig into the G29 yet.   But to make sure the bed was physically level, I wrote a small piece of code to probe (and measure) very close to the screws for the bed.   

Add this before the G28 code:
````
	case 27:	// Roxy Probe 3 points to manually level bed
	  {
      ProbeAction p1, p2, p3;
        p1 = ProbeDeploy; 
	p2 = ProbeStay; 
	p3 = ProbeStow;

      // Probe at 3 arbitrary points
      float z_at_pt_1, z_at_pt_2, z_at_pt_3;
	      
	SERIAL_PROTOCOLPGM("Outer Circle of Points:\n");

	z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, Z_RAISE_BEFORE_PROBING, p1, 4),
	z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, 4),
	z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, 4);

	SERIAL_PROTOCOLPGM("Middle Circle of Points:\n");

	z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X/1.75, ABL_PROBE_PT_1_Y/1.75, Z_RAISE_BEFORE_PROBING, p2, 4),
	z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X/1.75, ABL_PROBE_PT_2_Y/1.75, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, 4),
	z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X/1.75, ABL_PROBE_PT_3_Y/1.75, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, 4);

	SERIAL_PROTOCOLPGM("Inner Circle of Points:\n");

	z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X/3.0, ABL_PROBE_PT_1_Y/3.0, Z_RAISE_BEFORE_PROBING, p2, 4),
	z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X/3.0, ABL_PROBE_PT_2_Y/3.0, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, 4),
	z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X/3.0, ABL_PROBE_PT_3_Y/3.0, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p3, 4);

      clean_up_after_endstop_move();
//      set_bed_level_equation_3pts(z_at_pt_1, z_at_pt_2, z_at_pt_3);
	}
	break;
''''

and you may need to add this block of code to get the U shaped dip from happening:
````
#define MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	25

  static void do_blocking_move_to(float x, float y, float z) {
    float oldFeedRate = feedrate;

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (marlin_debug_flags & DEBUG_LEVELING) {
        print_xyz("do_blocking_move_to", x, y, z);
      }
    #endif

    #if ENABLED(DELTA)

      feedrate = XY_TRAVEL_SPEED;

 //
 // Should we break up the move because it is too big ???
 //

      if (abs(x-current_position[X_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(y-current_position[Y_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(z-current_position[Z_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE ) {

  		do_blocking_move_to( current_position[X_AXIS] + (x-current_position[X_AXIS])/2.0,
  				     current_position[Y_AXIS] + (y-current_position[Y_AXIS])/2.0,
  				     current_position[Z_AXIS] + (z-current_position[Z_AXIS])/2.0);
//
// At this point, we are halfway to where we want to be.  Do the other half of the move.
//

		do_blocking_move_to(x, y, z);
		feedrate = oldFeedRate;
		return;
      } else {		// No...  It is small enough to do in one piece.

      	destination[X_AXIS] = x;
      	destination[Y_AXIS] = y;
      	destination[Z_AXIS] = z;
      	prepare_move_raw(); // this will also set_current_to_destination
      	st_synchronize();
      }
    #else

      feedrate = homing_feedrate[Z_AXIS];

      current_position[Z_AXIS] = z;
      line_to_current_position();
      st_synchronize();

      feedrate = xy_travel_speed;

      current_position[X_AXIS] = x;
      current_position[Y_AXIS] = y;
      line_to_current_position();
      st_synchronize();
    #endif

    feedrate = oldFeedRate;
  }
''''

And in Configuration.h set the 3-Point leveling points to points very close to your set screws.  And then move them such that they are always turned on if you have Auto Bed Leveling turned on.

sarf2k4, 2015-10-14:
Hi, I think i had same problem before, the bed was tilted on the right side marlin registered as lower, left as higher, in the end during prints, the nozzle were scraping the bed on the right side. I decided to try cleaning my glass bed, put a new 3-4 layer of glue stick and test again, my problem somewhat gone.

Just my suggestion that might help

Sungod3000, 2015-10-14:
@sarf2k4 yes thats the pattern. Im using aluminium and I cleaned all the hairspray of at one time, but it didnt change anything.

sarf2k4, 2015-10-14:
I had the same last time when I switched to aluminium with inductive sensor, in the end I removed it because the nozzle isn't near enough. Not to mention that there's no way to see the 'current' z offset without lcd panel

The pattern I got before cleaning the glass were always descend at right side, ascend at left side, no matter how I tried to adjust the bed (once only) and limited to small prints only mostly. 

What about lowering your homing speed? I use around 45*4 I think. I saw in another issue topics saying lowering the speed would have higher accuracies

Sungod3000, 2015-10-14:
my homing speed is low already and the repeatability of the pattern is very good, so i dont doubt the accuracy of the sensor. the z offset show in the m502 read out btw.

i will try the old RichCattel version soon, and see if that gives me a different result

sarf2k4, 2015-10-16:
I ran into same problem last night, it really does adjusts incorrectly; having the nozzle raised on the left, lowered on the right. Now that I suspect there's something wrong with the mesh bed leveling being flipped making marlin produces this problem. 

For example in the physical world, the left side of the bed is lower and right side is higher, after the g29 routine, the marlin would register that on the left side is higher while right side is lower, thus during printing the nozzle are being pushed to scrape the bed surface at right side

This is just an assumptions, I attached an image showing the printer is printing on the first layer, originally the nozzle were too far; I used babystepping to lower the z axis, but the first layer were inconsistent

http://imgur.com/a/406Kf

After the print on the image above, I tried to physically raise the right side of the bed, probably by around 0.2-0.5, screw turned by about quarter counter clockwise. The results were the layer are somewhat uniform. 

Although I don't understand why, the print before the object in the image were also quite consistent too

I'm joining in this discussion to give my suggestions and opinion regarding this. Auto Bed Leveling aren't supposed to make a lot of problems

thinkyhead, 2016-02-11:
@Sungod3000 @sarf2k4 Are you still seeing this issue with the latest RC? If so, be sure to enable `DEBUG_LEVELING_FEATURE` and use `M115 S32` to turn on debugging for homing and leveling. Then you will get a detailed log when you do `G29`. That should provide a clue where it's messing up.

Sungod3000, 2016-02-11:
I disabled auto level but I remember that someone on the reprap forum suggested some verbose commands but they didnt yield any new information. Im making a new effector that will have space for my leveling probe, Ill give a new shot then.

sarf2k4, 2016-02-24:
Sorry I didn't notice you were replying to us, I probably solved it on my own printer, turns out to be adjusting the left and right motor did the trick (prusa i3 based) and make it leveled. But I still having same problem on my friend's printer, did the same but problem persist as well, I'm providing you the config.h, and verbose output for the bed at 0'c (idle),70'c and 110'c. The problem only occurred during at 70'c before; printing pla on bare glass without any gluestick. As of now after I insulated the heat to reach 110'c to print abs, it is good as far as I know (didn't do big prints yet on abs) and haven't been printing any pla ever since 2 days ago

Here is some [extra details](http://forums.reprap.org/read.php?262,629751) that you might need. Note that the one in the link is before I insulated the heatbed with cardboard

Bed at Idle:
`SENDING:G29 V4
G29 Auto Bed Leveling
Bed X: 15.000 Y: 20.000 Z: 2.466
Bed X: 100.000 Y: 20.000 Z: 2.837
Bed X: 185.000 Y: 20.000 Z: 3.015
Bed X: 185.000 Y: 97.000 Z: 2.846
Bed X: 100.000 Y: 97.000 Z: 2.711
Bed X: 15.000 Y: 97.000 Z: 2.102
Bed X: 15.000 Y: 174.000 Z: 2.089
Bed X: 100.000 Y: 174.000 Z: 2.256
Bed X: 185.000 Y: 174.000 Z: 2.840
Eqn coefficients: a: 0.00400850 b: -0.00245239 d: 2.41051340
Mean of sampled points: 2.57348108
planeNormal x: -0.004008 y: 0.002452 z: 1.000000

Bed Height Topography:
+-----------+
|...Back....|
|Left..Right|
|...Front...|
+-----------+
 -0.48481 -0.31732 +0.26642
 -0.47126 +0.13711 +0.27258
 -0.10796 +0.26396 +0.44130

Corrected Bed Height vs. Bed Topology:
 +0.17528 +0.00205 +0.24508
 +0.00000 +0.26765 +0.06240
 +0.17447 +0.20567 +0.04229

Bed Level Correction Matrix:
+0.999992 +0.000000 +0.004008
+0.000010 +0.999997 -0.002452
-0.004008 +0.002452 +0.999989`

Bed at 70'c
`SENDING:G29 V4
G29 Auto Bed Leveling
Bed X: 15.000 Y: 20.000 Z: 2.409
Bed X: 100.000 Y: 20.000 Z: 2.830
Bed X: 185.000 Y: 20.000 Z: 3.065
Bed X: 185.000 Y: 97.000 Z: 2.860
Bed X: 100.000 Y: 97.000 Z: 2.663
Bed X: 15.000 Y: 97.000 Z: 2.078
Bed X: 15.000 Y: 174.000 Z: 2.060
Bed X: 100.000 Y: 174.000 Z: 2.256
Bed X: 185.000 Y: 174.000 Z: 2.889
Eqn coefficients: a: 0.00444557 b: -0.00237775 d: 2.35381889
Mean of sampled points: 2.56773400
planeNormal x: -0.004446 y: 0.002378 z: 1.000000

Bed Height Topography:
+-----------+
|...Back....|
|Left..Right|
|...Front...|
+-----------+
 -0.50739 -0.31158 +0.32143
 -0.49015 +0.09483 +0.29187
 -0.15887 +0.26232 +0.49754

Corrected Bed Height vs. Bed Topology:
 +0.18206 +0.00000 +0.25514
 +0.01622 +0.22332 +0.04250
 +0.16441 +0.20773 +0.06508
[... 895 lines cut ...]
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H
`

thinkyhead, 2016-03-02:
What results do you get from `M48` "Repeatability Test". First thing is to make sure you have a reliable measurement device. If it's not very reliable, you may be able to slow down the probe and get better results.

sarf2k4, 2016-03-14:
Here is the m48 reports, I did 5 points. These are when the bed are cold

Front Left
Mean: -9.775862
Standard Deviation: 0.008708

Front Right
Mean: -9.939288
Standard Deviation: 0.005591

Back Left
Mean: -10.120197
Standard Deviation: 0.004413

Back Right
Mean: -10.012562
Standard Deviation: 0.007282

Center
Mean: -9.957020
Standard Deviation: 0.005061

Roxy-3DPrintBoard, 2016-03-14:
Those 'Standard Deviation' numbers look acceptable.  (They are not super good...  But they should be good enough)   You do have a fair amount of tilt to your bed.   The Auto Bed Leveling should be able to correct for that, but it would be wise to get the bed more physically level if you can.

thinkyhead, 2016-03-27:
Time to test with the latest `RCBugFix`. I suggest more adjustment, testing, adjustment, testing, and adjustment. I don't think we're seeing anything like a software issue here.

closed 2016-03-27 by thinkyhead