MarlinFirmware/Marlin #2666: second extruder error

coalman321, 2015-10-02:
i have changed my configs to fit my printer. but now i am adding a second extruder. i have checked configuration.h and pins.h but still get a compilation error when i up the extruder amount from 1 to 2.
here is said error.

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:665: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:665: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.168 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.168 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out
Exception in thread "Thread-3649" java.lang.IndexOutOfBoundsException: len=-7366
	at sun.font.FontDesignMetrics.charsWidth(FontDesignMetrics.java:501)
	at org.fife.ui.rsyntaxtextarea.TokenImpl.getWidthUpTo(TokenImpl.java:528)
	at org.fife.ui.rsyntaxtextarea.TokenImpl.getWidth(TokenImpl.java:499)
	at org.fife.ui.rsyntaxtextarea.RSyntaxUtilities.getTokenListWidth(RSyntaxUtilities.java:1142)
	at org.fife.ui.rsyntaxtextarea.RSyntaxUtilities.getTokenListWidth(RSyntaxUtilities.java:1121)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.getLineWidth(SyntaxView.java:304)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.calculateLongestLine(SyntaxView.java:110)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.updateMetrics(SyntaxView.java:892)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.setSize(SyntaxView.java:802)
	at javax.swing.plaf.basic.BasicTextUI$RootView.setSize(BasicTextUI.java:1722)
	at javax.swing.plaf.basic.BasicTextUI.getPreferredSize(BasicTextUI.java:912)
	at javax.swing.plaf.basic.BasicTextAreaUI.getPreferredSize(BasicTextAreaUI.java:120)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1662)
	at javax.swing.JTextArea.getPreferredSize(JTextArea.java:619)
	at javax.swing.JViewport.getViewSize(JViewport.java:999)
	at javax.swing.plaf.basic.BasicScrollPaneUI.syncScrollPaneWithViewport(BasicScrollPaneUI.java:278)
	at javax.swing.plaf.basic.BasicScrollPaneUI$Handler.stateChanged(BasicScrollPaneUI.java:1033)
	at javax.swing.JViewport.fireStateChanged(JViewport.java:1369)
	at javax.swing.JViewport.setView(JViewport.java:969)
	at javax.swing.JScrollPane.setViewportView(JScrollPane.java:1007)
	at org.fife.ui.rtextarea.RTextScrollPane.setViewportView(RTextScrollPane.java:253)
	at processing.app.Editor.setCode(Editor.java:1766)
	at processing.app.Sketch.setCurrentCode(Sketch.java:1000)
	at processing.app.Sketch.setCurrentCode(Sketch.java:980)
	at processing.app.Editor.statusError(Editor.java:2670)
	at processing.app.Editor$BuildHandler.run(Editor.java:1963)
	at java.lang.Thread.run(Thread.java:745)

Ant help at all would be greatly appreciated as this is a school project.
thanks.

Wackerbarth, 2015-10-02:
Well, you haven't supplied enough information. Since one of our ContinuousIntegration tests compiles with 2 extruders, we need to know just what you are attempting to compile.

Which version of the code are you compiling? What are you using to compile it? What do you have in your Configuration.h and also, if you changed anything, your Configuration_adv.h (Post it as a gist or push to some public repo)

Since the compiler is aborting, it is likely that there is a problem in one of your source files.

coalman321, 2015-10-02:
these are all the files i have changed in the firmware.
https://gist.github.com/6ac755b1ddce35de18c8.git

coalman321, 2015-10-02:
hmm not sure why but the gist link doesn't work. try this one.

https://gist.github.com/coalman321/6ac755b1ddce35de18c8

KiteLab, 2015-10-02:
This is a known error in release/1.0.x.
Please search for:
is:issue is:open POINTER_REGS 
if you want more reading:
is:issue POINTER_REGS 

The solution is ether to use an old arduino IDE 1.0.6,
or to use the RC-branch.

coalman321, 2015-10-02:
Can I still use my current configs?

coalman321, 2015-10-02:
Ok I did some of the suggested reading and found this error has been caused by an undefined temp sensor. However when this happens to mine, the second temp is defined.

KiteLab, 2015-10-02:
If you'll try Arduino 1.0.6 - yes, unchanged.
If you'll try the RC - no, not direct, but as a hint for the new config.
The fastest way to a new configuration is to make a diff against the original distributed config. Then correct the new config top down, making the diff as done. Positions have been shifted an options renamed - but you'll find everyting.

coalman321, 2015-10-02:
so what version is the current RC? also if i update to the RC will  i have to backdate my arduino client to 1.0.6?

KiteLab, 2015-10-02:
https://github.com/MarlinFirmware/Marlin/tree/RC
Either or.

coalman321, 2015-10-02:
ok updated and it now works with 2 extruders thanks.

closed 2015-10-02 by coalman321