MarlinFirmware/Marlin #2672: Problems with auto bed leveling and lcd

orrin-naylor, 2015-10-04:
Hay guys sorry if this is the wrong place for this I'm kinda a newbie.
I am running the newest version of marlin on a RAMPS 1.4 board. When I have auto bed leveling and have my discount LCD controller defined the board crashes and nothing works. If I disable one than everything works good. It is only when they are both enabled that nothing works.
Thanks for any help.

AnHardt, 2015-10-05:
Please give us some more hints.
What branch/version are you on?
What "discount LCD controller" do you have? REPRAP_DISCOUNT_SMART_CONTROLLER is a charecter display. REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is a graphical display.
If you have a (rare, old) Arduino Mega 1280 you may be out of memory if you defined in some other memory hogs.
In most cases a link to your 'Configuration.h' will tell us all we need to know to help you.

pbrgle1, 2015-10-09:
try taking the 5V for the sevro from an outside power source. i had the same problem (i have Megatronics 3.0) and found the solution here - http://reprapworld.com/documentation/mtQSv3.pdf

'' Note you can also draw 5v from a different power supply, in case you have more peripherals
like LCD drawing power. ''

closed 2015-11-03 by AnHardt