MarlinFirmware/Marlin #2673: Hotend temp control and display

ruggb, 2015-10-04:
I was having some problems here with RC2 91b515...
after much struggling I reverted back to 1.0.2-1 and immediately found this ?BUG?
(not part of my problem)
when running PID tune on RC2 for E1 the temp displays on E0 bar unless I change to Extruder 2
after PID finishes it shows on E1 bar.
graph shows E1 regardless of which Extruder is selected.
doing the same in 1.0.2 shows temp on E1 bar during PID run and after
graph shows E0 unless I chg to Extruder 2.
IMHO 1.0.2-1 is the better way though I was confused initially and had to ask the question but that was because I didn't know what that block was for.
RC2 is actually showing false info. I.E., E1 temp in E0 bar.

Since I started writing this I found and fixed MY problem. It was fan motor noise being picked up on the thermistor lines. Drove it crazy.

AnHardt, 2015-10-05:
What means "bar" here?
On a display? If yes - graphical or character?
"graph"?  On a host? If host - what host?
"block"? What block?

ruggb, 2015-10-05:
sorry - using Repetier-Host
1 bar for each extruder showing temp
the block refers to the extruder selection block on the left of Repetier display
It would be nice if Repetier would split the display when 2 extruders were configured, but they don't

But showing E1 temp on E0 bar is obviously a Marlin function because it works dif in the release vs RC2 on the same ver of Repetier

AnHardt, 2015-10-05:
So we have to take a look on the output of M105.

ruggb, 2015-10-06:
u obviously have the correct info. this issue is sending it with the right ID. It is correct in Release ver. 
What would be great is if that info could somehow get to the graph without having to change the extruder box. Technically it would be wrong info since the box might show the right or wrong extruder selected.
I assume Repetier selects the data for the graph so u would send the data 2x, one for E0 and one for E1. That would make both temp bars show the same thing, but I see that as a lesser evil.
just MHO.

AnHardt, 2015-11-10:
@ruggb 
Please download https://github.com/AnHardt/Marlin/archive/M105andCo.zip, apply your config and test M303 with both extruders again. The output should be improved.
More info about the patch https://github.com/MarlinFirmware/Marlin/pull/2768

thinkyhead, 2016-02-24:
How is this working now with the current `RCBugFix`? Have we patched everything that needs it? I see that #2822 replaced #2768 and was merged, but #2781 is still open. If #2781 looks okay to everyone, perhaps that can be merged for RC4 also.

AnHardt, 2016-02-24:
`print_heaterstates()` is in and used by M105, M109, M190, and M303 and is working as expected, in RCBugFix. So everything seems to be fine to __close__ #2781 and this one.

Roxy-3DPrintBoard, 2016-02-24:
> PS. 
> There are still gaps in the temperature graphs during G29, M48, M0/1 (and maybe other long lasting commands), but there temperatures and heater states are less interesting.

We may want to call idle() during the long probing loops.   It wouldn't hurt G29 or M48 to do that.

AnHardt, 2016-02-24:
Dear Roxy
How does the host know the temperatures of the printer?
a. By sending M105 and evaluating the response.
b. The tested hosts (OctoPrint, RepetieHost, Pronterface) also understand a string formatted like the response of M105, without 'ok' in front, when sent unasked.

In M109, M190, M303 we used to send information, about the temperatures and heater-states, in our own random, not understood by the hosts, format. This output is now changed to the M105 format.

Idling does not help! Indeed we already are idling in G29 (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L3000). As long as, for example, G29 is busy, no new M105 can be executed. We could actively send a M105 response - but that would mess up G29s output. __I__ don't want that.

Roxy-3DPrintBoard, 2016-02-24:
I got confused...  I was thinking Marlin wasn't controlling the temperature within a tight enough band.   And calling idle() would help that.   I didn't realize we were talking about the graphs that get plotted on the computer screen.

AnHardt, 2016-02-24:
As far as i know there are no gaps in controlling the temperatures in Marlin.

ruggb, 2016-03-01:
I did a quick test on 9a3408e:
the comm errors I was seeing appear to have stopped - not sure if u addressed that specifically or it was a result of another fixed issue. 
The issue with the misguided temps and chart with the M303 command are not back - it was fixed in a previous bug fix.

closed 2016-03-02 by thinkyhead