MarlinFirmware/Marlin #2674: Y endstop problem on Printrbot Simple Metal

dattas, 2015-10-05:
I'm installing Marlin a printrbot simple metal, using 1.1.0-RC2 here is my configuration.h: https://gist.github.com/dattas/2287cd49da92c6dcdb26

It appears as though the firmware disables the Y endstop, I don't have `DISABLE_MAX_ENDSTOPS` or `DISABLE_MIN_ENDSTOPS` defined as I have 2 min endstops (x and z) and 1 max endstop (y). I also have `Y_HOME_DIR 1` defined to ensure it goes to the Y MAX when homing. This configuration worked as expected on an older version of Marlin before I went through and upgraded things for the latest RC.

When I use M119, it always shows the Y endstop as triggered, no matter what state it is actually in. If it is actually being triggered or not it; shows triggered, which makes me think it is being disabled somehow within the firmware. Reverting to an old firmware fixes the issue.

AnHardt, 2015-10-05:
It may be worth to play with the ENDSTOPPULLUPS. Especially if the endstop has an other hardware then the others, or is connected a different way. (If i remember correct, the default may have changed here.)

dattas, 2015-10-05:
The hardware is the same and is connected in the same way. I can remove the define for the `ENDSTOPPULLUPS` and try to manually just define the 3 ones that will be pulled up but if I'm not mistaken, by default `ENDSTOPPULLUPS` just pulls them all up if that is defined so I would doubt changing from the global to the individual would do anything. I'm guessing that somewhere in the pins or somewhere else the pin is getting set to -1 or something similar.

AnHardt, 2015-10-07:
Your "printrbot simple metal" has a RAMPS mainboard?

dattas, 2015-10-07:
@AnHardt Nope, it has a printrboard or `BOARD_PRINTRBOARD` as configured in my configuration file linked above. I've been playing with the `ENDSTOPPULLUPS` and so far nothing has resolved the issue with the Y MAX endstop being wrongly disabled.

AnHardt, 2015-10-07:
Sorry overlooked you define before the #ifndef MOTHERBOARD.

Can this have anything to do with your problem?
```
#if ENABLED(SDSUPPORT)
  #define Y_STOP_PIN       37 // Move Ystop to Estop socket
#else
  #define Y_STOP_PIN        8 // Ystop in Ystop socket
#endif
```

If i remember correctly this was done to support sd-readers on a external panel - not at the board itself.

dattas, 2015-10-07:
Looks like you hit the nail on the head, when I comment out `SDSUPPORT` the Y max endstop starts working as it has in the past. I hate to lose my sd printing, any chance that it can be re-looked at? Maybe not change the Y axis pin if no external panels are defined?

AnHardt, 2015-10-07:
See updated comment above.

AnHardt, 2015-10-09:
@dattas 
Feedback please.

dattas, 2015-10-10:
So using the estop works, however since there is an SD card reader on the board, I feel like it is quite silly to force people to rewire their stock printrboards to use stock features. My recommendation stands, if there is no external panel defined, there should be absolutely no reason to remap the Y endstop to another pin.

KiteLab, 2015-10-10:
Is your on board cardreader working now?
If so, it's a problem with the boars design - not Marlin.
If no, it's a problem with the pin file for your board.

Wackerbarth, 2015-10-10:
@dattas -- No one is required to change their wiring. However, they MUST choose the appropriate software definitions (pins file) to represent the way that the wires are connected. I liken your situation to that of someone saying "I don't understand why the radio in my BMW doesn't work properly. I'm following the instructions in the Toyota owners manual."

KiteLab, 2015-10-11:
#2418 seems to be a good entry point to read why it is as it is.

closed 2015-11-08 by AnHardt