MarlinFirmware/Marlin #2675: Graphical displays can not display negative X and Y coordinates

AnHardt, 2015-10-05:
Graphical displays can not display negative X and Y coordinates because they use `ftostr31ns()` to convert.
The `ns` means 'no sign' here.
Not so nice if you have a delta.

AnHardt, 2015-10-05:
Will return to this when iss2661 is fixed - same code area.

AnHardt, 2015-10-06:
https://github.com/MarlinFirmware/MarlinDev/pull/181
is related/similar but for the C-LCD's

AnHardt, 2015-10-27:
We have space for 5 chars. We want to display positive and negative numbers with up to 3 digits before the dot and 1 behind. With sign and dot we'd need 6 chars. We could drop the fraction for numbers < -99.9
```
12345

-999.
-99.9
 -9.9
 -0.1
  0.0
  0.1
  9.9
 99.9
999.9 
```

Looks like a plan.

thinkyhead, 2016-02-10:
Do we have a patch for this yet? Seems like it should be a quick one. Still waiting for #2661?

closed 2016-04-23 by jbrazio