MarlinFirmware/Marlin #2677: Printer stops, when start to heating extruder

ohrie, 2015-10-06:
Hey there!

I'm using the RC branch of Marlin. But even in development branch, i get the same error. A couple of seconds after i started heating my one and only extruder, in the console this show's up:

< 20:39:19: Error:Heating failed, system stopped! Heater_ID: 0
< 20:39:19: Error:Printer halted. kill() called!

The log of the console, before i got this error, is on the end of this issue.
I have a Rigidbot 3D & i'm very frustated, because my printer worked never before.

  20:38:42: Attempting to connect to printer
  20:38:43: Connection opened
< 20:38:52: °start
> 20:38:52: N1 M110 *2
< 20:38:52: echo:Marlin 1.1.0-RC2
> 20:38:52: N2 M115 *4
< 20:38:52: echo: Last Updated: 2015-09-28 12:00 | Author: (none, default config)
< 20:38:52: Compiled: Oct  6 2015
> 20:38:52: N4 M111 S6 *67
< 20:38:52: echo: Free Memory: 4811  PlannerBufferBytes: 1232
< 20:38:52: echo:Hardcoded Default Settings Loaded
< 20:38:52: echo:Steps per unit:
< 20:38:52: echo:  M92 X44.31 Y22.15 Z1600.00 E53.50
< 20:38:52: echo:Maximum feedrates (mm/s):
< 20:38:52: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
< 20:38:52: echo:Maximum Acceleration (mm/s2):
< 20:38:52: echo:  M201 X800 Y800 Z100 E10000
< 20:38:52: echo:Accelerations: P=printing, R=retract and T=travel
< 20:38:52: echo:  M204 P600.00 R1000.00 T3000.00
< 20:38:52: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
< 20:38:52: echo:  M205 S0.00 T0.00 B20000 X8.00 Z0.40 E5.00
< 20:38:52: echo:Home offset (mm):
< 20:38:52: echo:  M206 X0.00 Y0.00 Z0.00
< 20:38:52: echo:Material heatup parameters:
< 20:38:52: echo:  M145 M0 H180 B70 F255
< 20:38:52: echo:  M145 M1 H240 B110 F255
< 20:38:52: echo:PID settings:
< 20:38:52: echo:  M301 P9.00 I0.15 D85.00 C100.00 L20
< 20:38:52: echo:Filament settings: Disabled
< 20:38:52: echo:  M200 D3.00
< 20:38:52: echo:  M200 D0
> 20:38:56: N5 M111 S6 *66
< 20:38:57: FIRMWARE_NAME:Marlin 1.1.0-RC2 From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
> 20:39:01: N12 M106 S76 *69
> 20:39:03: N15 M104 S200 *115

What's wrong?

AnHardt, 2015-10-06:
Cite from 'Configuration_adv.h'
```
  /**
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   */
  #define WATCH_TEMP_PERIOD 16                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
```

Look on your display or host when heating begins - is the temperature rising at all?
If you can see a rising temperature increase WATCH_TEMP_PERIOD and or decrease WATCH_TEMP_INCREASE. 
If you end up with very different values, probably your heater or thermistor is defekt

ohrie, 2015-10-07:
Thanks for this advice!
I will try it :)

AnHardt, 2015-10-09:
Please report back - or simply close this issue.

closed 2015-10-09 by ohrie