MarlinFirmware/Marlin #2679: Delta Printer moves to illegal coordinates

boulway, 2015-10-07:
Hi there,

I was a silent reader for a long time but I guess I should contribute a bit too.
So here I go.

My Setup is: Rostock Mini Pro with Sanguinololu / Atmega1284P

I had a lot of trouble getting my Delta printer to move the way it should and after some tests I figured out, that the Delta is moving to illegal coordinates.
For example:
-	Send G1 Z90 X80 Y80 to drive to one edge of the print bed
-	Printer moves arms correctly until one of the arms reaches its maximum (90° to one smoothrod)
-	Now the printer glitches and it moves all axis equally down like it got just a Z decrement instruction
-	Of course the head crashes if you cannot reset the printer fast enough ;)

I guess the Software glichtes if driving to coordinates that are in an illegal section of the print bed.
For my I fixed it with manually testing how far I can to one edge until the system will move down till head crash.
I hardcoded it in the software now but it would be nice if there could be a test to avoid the automatic head crashing.

Best Regards,
Constantin

clefranc, 2015-10-07:
@boulway Resurrecting an old issue, I like that.

https://github.com/RichCattell/Marlin/issues/7
https://github.com/MarlinFirmware/Marlin/issues/1362
http://forums.reprap.org/read.php?267,355792
https://groups.google.com/forum/#!topic/deltabot/tUJ1ty6yebg

At the time, I've been told to not go to the *forbidden* coordinates. I think it's now time to fix this. 

@Roxy-3DPrintBoard @AnHardt What do you think about it?

clefranc, 2015-10-07:
@ProjectEnder
Can we have more information on how you solved this issue in your configuration: https://github.com/MarlinFirmware/Marlin/issues/1362#issuecomment-89040817

Roxy-3DPrintBoard, 2015-10-07:
So, I'm not sure about this...  But I've been seeing almost the opposite condition.  I am not sure if it is related but my Probable area is set to the Printable area minus 10 mm.    I wanted the probable area to be bigger so I bumped up the printable size with the intent of just not going off the bed.  (I knew this wasn't legit but it was an easy way to make the probable area bigger while I tried different things and started adjusting things like the DELTA_RADIUS numbers)

Well...  If you move the nozzle some where that one of the towers goes past 90 degrees the printer goes nuts.  And all of the carriages start moving together.  But in my case they all started moving up together.   

I think you can force this behavior by doing a move where a carriage angle goes past 90 degrees.

clefranc, 2015-10-07:
> in my case they all started moving up together

Same behavior here.

I think XY restrictions can be safely computed in `clamp_to_software_endstops()`, to circumscribe coordinates inside the printable radius.

AnHardt, 2015-10-07:
Clamping to a radius means solving a sqrt. 
The shape of the reachable area is not necessary a circle - i suppose more like the shape of the piston of a wankel engine.
In an other attempt wee could try to check the horizontal angel of the diagonal rod, but as far i know we'd have to calculate them. Trig. functions are not cheap.

Checking if the `carriageHight <= (tipHeight + Nozzel-Joint_Offset)` is easy to calculate - but we have to find a place where to apply it and what to do then.

Roxy-3DPrintBoard, 2015-10-07:
> Clamping to a radius means solving a sqrt. 

Is this true?   Could we clamp to a radius^2 and never do the square root?

AnHardt, 2015-10-07:
@Roxy-3DPrintBoard 
You are right.
x*x + y*y <= R^R - where R^R can be precalculated.
Remains the problem that this will probably clamp to much.

clefranc, 2015-10-07:
>The shape of the reachable area is not necessary a circle

You're right, but most (all?) delta use a round bed, if not, the 90° still apply to them. So, we have a safe margin we can apply exceeding the radius before the clamp kicks in.

Roxy-3DPrintBoard, 2015-10-07:
> Remains the problem that this will probably clamp to much.

Can you help me understand when it clamps too much and what the shape of the clamped area should be?   And to the point of the 90 degrees applies to everybody,  Is going to 90 degrees OK?  Or do we need to back off of that a few degrees?   The 90 degrees shouldn't be a problem.  Rather, it is the fact there is no precision at that angle, right?

clefranc, 2015-10-07:
@Roxy-3DPrintBoard See http://www.reprap.org/mediawiki/images/b/b5/Rostock_Delta_Kinematics_3.pdf for delta printable area and error near towers.

Roxy-3DPrintBoard, 2015-10-07:
Thank You!  That is a very good link!  I'm going to read the whole thing before I go any further trying to calibrate my printer!

clefranc, 2015-10-07:
An economic solution to add to the current clamping routine:

[image]

boulway, 2015-10-07:
You guys are absolutely crazy. :)
I wish my devs at work would think as fast as you...

Chrsitans picture nails it!
At the moment I lowered the current quadratic software clamps to fit inside the circled area.
from a 200x200mm printbed it leaves 108x108mm at the moment. But I think it could be extended to 160x160 with the circular software clamp area.

clefranc, 2015-10-07:
OK, I've tested the circle method, it works (a circle with flat side, or a rounded square, your choice). But another problem arise: the host.

The host is only aware of square limits, its controls can't track the new clamp shape. So the firmware must send the coordinates (at least with Repetier-Host) to get the current clamped position updated.

Another problem is the `lcd_move_x()` & `lcd_move_y()` in the LCD routines. They don't use `clamp_to_software_endstops()`.

AnHardt, 2015-10-07:
Thank you @clefranc .
As supposed - like the piston of a wankel engin (https://upload.wikimedia.org/wikipedia/commons/thumb/f/fc/Wankel_Cycle_anim_en.gif/220px-Wankel_Cycle_anim_en.gif)

So a circle is OK.

@Roxy-3DPrintBoard 
90° is a hard limit. At 90° you have the highest possible distance from the tower. Over 90° the distance is shrinking again. The kinematic equations have no solution for higher distances. So 90° is ok, but not one step further.

Mhhh. Looks as if my bed is to large or my diagonal rots to short - cant reach the full bed anyway. (Sintron Mini Kossel) 

Let's look for the CLIP_RADIUS.
It's the DELTA_PRINTABLE_RADIUS,
Can we calculate this?
It must be smaller then the beds radius.
It must be smaller then DELTA_DIAGONAL_ROD + DELTA_EFFECTOR_OFFSET + DELTA_CARRIAGE_OFFSET - DELTA_SMOOTH_ROD_OFFSET. That's our maximum reach at 90°.
It must be smaller then DELTA_SMOOTH_ROD_OFFSET - DELTA_CARRIAGE_OFFSET  - DELTA_EFFECTOR_OFFSET. (That's a collision with a tower)
A diagonal rod - carriage collision is difficult to calculate.

clefranc, 2015-10-07:
I'm testing this right now:
```cpp
void clamp_to_software_endstops(float target[3]) {
#if ENABLED(DELTA)
  bool delta_clamp = sq(target[X_AXIS]) + sq(target[Y_AXIS]) >= sq(DELTA_PRINTABLE_RADIUS + 10);
#endif
  if (min_software_endstops) {
#if ENABLED(DELTA)
    if (delta_clamp) {
      set_destination_to_current();
      SERIAL_PROTOCOL("Software clamp detected X:");
      SERIAL_PROTOCOL(current_position[X_AXIS]);
      SERIAL_PROTOCOL(" Y:");
      SERIAL_PROTOCOL(current_position[Y_AXIS]);
      SERIAL_EOL;
    }
#endif
    NOLESS(target[X_AXIS], min_pos[X_AXIS]);
    NOLESS(target[Y_AXIS], min_pos[Y_AXIS]);
    float negative_z_offset = 0;
#if ENABLED(AUTO_BED_LEVELING_FEATURE)
    if (zprobe_zoffset < 0) negative_z_offset += zprobe_zoffset;
    if (home_offset[Z_AXIS] < 0) {
#if ENABLED(DEBUG_LEVELING_FEATURE)
      if (marlin_debug_flags & DEBUG_LEVELING) {
        SERIAL_ECHOPAIR("> clamp_to_software_endstops > Add home_offset[Z_AXIS]:", home_offset[Z_AXIS]);
        SERIAL_EOL;
      }
#endif
      negative_z_offset += home_offset[Z_AXIS];
    }
#endif
    NOLESS(target[Z_AXIS], min_pos[Z_AXIS] + negative_z_offset);
  }
  if (max_software_endstops) {
#if ENABLED(DELTA)
    if (delta_clamp) {
      set_destination_to_current();
      SERIAL_PROTOCOL("Software clamp detected X:");
      SERIAL_PROTOCOL(current_position[X_AXIS]);
      SERIAL_PROTOCOL(" Y:");
      SERIAL_PROTOCOL(current_position[Y_AXIS]);
      SERIAL_EOL;
    }
#endif
    NOMORE(target[X_AXIS], max_pos[X_AXIS]);
    NOMORE(target[Y_AXIS], max_pos[Y_AXIS]);
    NOMORE(target[Z_AXIS], max_pos[Z_AXIS]);
  }
}
```

AnHardt, 2015-10-07:
@clefranc 
That looks complicated. Better then allowin to far moves and trowing an error is to prevent to large input.

```
static void lcd_move_x() { _lcd_move(PSTR(MSG_MOVE_X), X_AXIS, X_MIN_POS, X_MAX_POS); }
static void lcd_move_y() { _lcd_move(PSTR(MSG_MOVE_Y), Y_AXIS, Y_MIN_POS, Y_MAX_POS); }
static void lcd_move_z() { _lcd_move(PSTR(MSG_MOVE_Z), Z_AXIS, Z_MIN_POS, Z_MAX_POS); }
```

We can only move one axis at once - so we can limit the x-input depending on y and the other way around.

clefranc, 2015-10-07:
> That looks complicated.

What's looks complicated?

Roxy-3DPrintBoard, 2015-10-07:
@AnHardt

> @Roxy-3DPrintBoard 
> 90° is a hard limit. At 90° you have the highest possible distance from the tower. Over 90° the distance is shrinking again. The kinematic equations have no solution for higher distances. So 90° is ok, but not one step further.

Right.  I understand the diagonal rods reach the furthest out at 90 degrees.  I was more concerned that we are going to be multiplying by floating point numbers that are very close to zero ( cos(0) ) and thus lose a lot of precision when we add or subtract them.   I haven't reviewed the actual steps the calculations take so I don't know if this is really an issue or not.

AnHardt, 2015-10-07:
Throwing errors. But my thoughts have been at LCD-menus when the message poped up. 

Did clamp_to_software_endstops() really fail silent - apparently yes. 
If this is an error we should correct that - if not we don't need to throw errors for the deltas.

AnHardt, 2015-10-07:
For Pronterface and Repetier-Host you can set up circular beds. In Repetier the bed is then defined by a radius. In Pronterface even the round bed looks as if defined by x/y. In Pronterface you have an extra option to clip manual moves. For Repetier i know the manual moves are chipped for the rectangular beds.
I did not test what really is be clipped - no delta. (And if i write all day long, that never will be finished)
If the hosts do not chip for a radius or rounded square, it's worth to make feature requests or bug reports.

AnHardt, 2015-10-07:
@Roxy-3DPrintBoard  @clefranc 
Thank you for this very constructive afternoon / evening. 
As a bonbon i have little OpenScad animation of a moving delta for you.
https://www.youtube.com/watch?v=FgfdqRlcbAI&feature=youtu.be
You can find the source at
https://gist.github.com/AnHardt/235703d24addb7a4086e

Are we still bug fixing or will this grow to a new feature.

Roxy-3DPrintBoard, 2015-10-07:
My vote is we treat as both a bug and a new feature.   Let's get a few more opinions.

clefranc, 2015-10-07:
> we don't need to throw errors for the deltas

This is not an error, just the firmware reporting back the position of the clamp, so the host is *aware* of the location of the nozzle. It's not supposed to happen during a print, only when moving the effector manually. It helps maintain the coordinate in sync. If not, the host will continue display movement as if nothing happened until it clamp its programmed value in its settings (Repetier-Host).

Pronterface doesn't track the location of the nozzle, so the reporting is ignored. Not tried other host...

Regarding the LCD, there is nothing to report back, just stop the move when the nozzle comes outside the clamp area.

Roxy-3DPrintBoard, 2015-10-07:
@Wackerbarth
OK.  My Mistake!  I was thinking that without the clamp, if the nozzle moved to a bad place it would lose position because its travel was blocked.   I think the MarlinDev tree is able to accept updates now.   Let's see if Wackerbarth confirms that.

AnHardt, 2015-10-07:
He menu part could look like
```
#if ENABLED(DELTA)
  static float delta_clip_radius_2 =  DELTA_CLIP_RADIUS * DELTA_CLIP_RADIUS;
  static int delta_clip( float a ) { sqrt(delta_clip_radius_2 - a*a); }
  static void lcd_move_x() { _lcd_move(PSTR(MSG_MOVE_X), X_AXIS, -delta_clip(current_position[Y_AXIS]), delta_clip(current_position[Y_AXIS])); }
  static void lcd_move_y() { _lcd_move(PSTR(MSG_MOVE_Y), Y_AXIS, -delta_clip(current_position[X_AXIS]), delta_clip(current_position[X_AXIS])); }
#else
  static void lcd_move_x() { _lcd_move(PSTR(MSG_MOVE_X), X_AXIS, X_MIN_POS, X_MAX_POS); }
  static void lcd_move_y() { _lcd_move(PSTR(MSG_MOVE_Y), Y_AXIS, Y_MIN_POS, Y_MAX_POS); }
#endif
```

AnHardt, 2015-10-07:
@clefranc 
What makes delta clipping different from cartesian clipping, so that one requires a report but not the other?
The host programs can not evaluate the message in the current fom. Useful for a user reading the terminal output - yes.
The hosts are not prepared to receive such a message. To make parsing simpler for the hosts we could use the output format of M114. But even then they are not prepared to evaluate the message if they have not send the M114 themselves.

Or we could use the format of the report the hardware endstpos give ( checkHitEndstops() ). But i doubt the host react on that.

AnHardt, 2015-10-07:
If you realy have a square bed
```
#if ENABLED(DELTA)
  static float delta_clip_radius_2 =  DELTA_CLIP_RADIUS * DELTA_CLIP_RADIUS;
  static int delta_clip( float a ) { sqrt(delta_clip_radius_2 - a*a); }
  static void lcd_move_x() { _lcd_move(PSTR(MSG_MOVE_X), X_AXIS, -min(X_MIN_POS, delta_clip(current_position[Y_AXIS])), min(X_MAX_POS, delta_clip(current_position[Y_AXIS]))); }
  static void lcd_move_y() { _lcd_move(PSTR(MSG_MOVE_Y), Y_AXIS, -min(Y_MIN_POS, delta_clip(current_position[X_AXIS])), min(Y_MAX_POS, delta_clip(current_position[X_AXIS]))); }
#else
  static void lcd_move_x() { _lcd_move(PSTR(MSG_MOVE_X), X_AXIS, X_MIN_POS, X_MAX_POS); }
  static void lcd_move_y() { _lcd_move(PSTR(MSG_MOVE_Y), Y_AXIS, Y_MIN_POS, Y_MAX_POS); }
#endif
```

clefranc, 2015-10-08:
> What makes delta clipping different from cartesian clipping.

From the firmware point of view, nothing. With our implementation, the firmware ignore the move in delta or cartesian. From a host point of view, it's a little different. The host (in this case RH) have parameters to be aware of the clamp, but even with a round bed, the square limits are still used (simple to evaluate). You end up with delta coordinates that are not in sync.

> The host programs can not evaluate the message in the current fom.

This is the RH trick: if the firmware send coordinates to RH in the form of X:x.x, it is automagically updated in its controls. This *hidden* feature made me pull hair on my head when I developed a G132 command (I was using coordinates report via serial). I think this is the main reason @thinkyhead use debug reports without X, Y or Z explicitly sent in the serial stream.

AnHardt, 2015-10-08:
Ahhh! OK. :+1:

clefranc, 2015-10-08:
@AnHardt Can you safely replace `DELTA_CLIP_RADIUS` by `DELTA_PRINTABLE_RADIUS`? It'll prevent to introduce another `#define`. The `_MIN_POS` & `_MAX_POS` are based on the later anyway.

AnHardt, 2015-10-08:
`DELTA_PRINTABLE_RADIUS` is in all 4 cases a number the user has to put in. Yes, you can replace `DELTA_CLIP_RADIUS` with `DELTA_PRINTABLE_RADIUS`.

The calculation in 
https://github.com/AnHardt/Marlin/pull/3  Mini Kossel
is a bit more sophisticated (if it works).

Even the `DELTA_PROBABLE_RADIUS` can be calculated with `DELTA_PRINTABLE_RADIUS - sqrt(X_PROBE_OFFSET_FROM_EXTRUDER * X_PROBE_OFFSET_FROM_EXTRUDER  +  Y_PROBE_OFFSET_FROM_EXTRUDER * Y_PROBE_OFFSET_FROM_EXTRUDER)`

Wackerbarth, 2015-10-08:
@Roxy-3DPrintBoard -- Yes, MarlinDev should be fine for new work.

I would like to have us try a different workflow. I'll explain it in a separate post.
But, in keeping with that workflow, I have created a "DeltaMotionLimits" branch for this issue.
At the present time, it is the same as "dev" and ready for pull requests.

AnHardt, 2015-10-08:
@clefranc 
In my branch we talked about the sense of DELTA_CLIP_RADIUS.

If we clip to inside DELTA_PRINTABLE_RADIUS and calculate X_MIN_POS, X_MAX_POS, Y_MIN_POS, Y_MAX_POS  from the same value we'll ever get a perfect circle, not a rounded square.

AnHardt, 2015-10-08:
I'll give the "DeltaMotionLimits" branch a chance - even if i don't like the idea of that scheme at all - and even if i'm still in bug-fix mode and MarlinDev sounds like a feature.

https://github.com/MarlinFirmware/MarlinDev/pull/235
https://github.com/MarlinFirmware/MarlinDev/pull/236

Well, at least i tried it. :-(

clefranc, 2015-10-08:
Why not use it in _lcd_move()?

clefranc, 2015-10-08:
Using `sqrt` in the LCD have no impact on a running print. This is why I've used your little formula
`x^2 + y^2 >= R^2` in `clamp_to_software_endstops`, to not introduce others `sqrt` in the print loop.

Are you going to propose something to limit coordinates in `Marlin_main.cpp` as well?

AnHardt, 2015-10-08:
In _lcd_move() i could not simply take the other coordinate - but have to make several cases.

The code in `Marlin_main.cpp` is your baby.

AnHardt, 2015-10-13:
@clefranc 
Did you give up here? I have seen well looking code in your 'Marlin4Delta'.

AnHardt, 2015-10-15:
@clefranc 
Is it the environment of MarlinDev?
It's a bug! We could try to fix it in `RCBugFix` or a fork of that.

Roxy-3DPrintBoard, 2015-10-15:
> Is it the environment of MarlinDev? (I don't like it very much)

Are you speaking about RC2 ?    Is there a problem with RC2?   It looks just like the code used to look before the Astyle changes.

clefranc, 2015-10-15:
@AnHardt My fork is a mix of pre-post Richard's "new system" MarlinDev.

Since I've a round bed, I didn't bother to increase the diameter for quared ones (in my fork). I think the use of a new variable (like you suggested) and an offset is more suitable, like in my code example above (`DELTA_PRINTABLE_RADIUS + 10`), i.e.:
```cpp
#define DELTA_CLIP_RADIUS = DELTA_PRINTABLE_RADIUS + offset // clamp area in mm for delta printer
```
The `offset` will give square bed more printable area.

Unless someone have a more "precise" calculation for detecting 90° angle, that don't increase lag for deltas, you can produce a PR, I'll not do it, my fork is too mess up, I don't have time to rebase it (if even possible).

I'm also switching to 32bit, and for other reasons, will not participate much in this repo.

Roxy-3DPrintBoard, 2015-10-15:
> Is it the environment of MarlinDev? (I don't like it very much)\

> I'm also switching to 32bit, and for other reasons, will not participate much in this repo.

Can we start fresh with a constructive conversation about what the head of the Development Branch needs to look like?   Probably this isn't the correct thread to have this discussion.  But where ever we can have the discussion is OK with me.

AnHardt, 2015-10-15:
@Roxy-3DPrintBoard 
No. The Marlin repository is just the way i like it.
But setting up the MarlinDev environment is, until now, a pain for me. Maybe because i'm a old mad dog and dislike fast transformation of my environment at all. Discovering the not working Travis-check in the morning frustrated me even more. I'll just give it a bit more time to mature. For now the RC/RCBugFix- playgrond, with all the bugs, is large enough for me and a dozen other programmers (we don't have). Until i can concentrate on making features again, a lot of time will go.

Politics.
I try to avid making politics, or giving directions for the further development her. That's much to time consuming in a florin language. When i tried i always had the impression to be far far off. Please excuse my occasional burst outs, then my mood is cooking, most times because something prevents me to do real work.

Time.
At least from Sa. to the Su. one week later i'll be offline. Please have a occasional look on the issue list. Please hit on @Wackerbarth's fingers when he tries the next revolution, like when @thinkyhead was moving

AnHardt, 2015-11-07:
There is an further clipping problem around.
Right after we homed to the top end-switches we can not move on x or y direction. For that one of the carriages would have to go up. The forbidden area can be roughly described by 3 spheres, with the center at the middle of the carriage-diagonal_rod joints when the carriage is homed, with a radius of the distance from the joints to the nozzles tip. The spheres have one common point at (0.0,zmax).

Roxy-3DPrintBoard, 2015-11-07:
Yes...  I didn't realize it until you explained it.   

> The forbidden area can be roughly described by 3 spheres, with the center at the middle of the carriage-diagonal_rod joints when the carriage is homed

That is going to be a very goofy looking shape.   And my guess is it would be very expensive computationally to watch for.

Wackerbarth, 2015-11-07:
For a delta, the min limit is Z=0 and the max limits should be checked in the actuator space rather than the object space.

thinkyhead, 2016-02-19:
One way I tried to approach this — and it may work, sort-of — was to allow any XYZ coordinates to be set, but then limit the positions of the carriages by constraining the angle of the delta arms so they can never be straight up-down. The easiest way is to examine at the triangle and get the ratio between the vertical side's length (effector Z minus carriage Z) and the horizontal side's length. When the ratio is very large, meaning the arm is nearly vertical, don't allow it.

That can be simplified by just limiting the allowed distance between effector Z and carriage Z to something like 80% of the arm length.

This approach avoids doing any extra `sqrt` calculations, but the down-side is that it allows the physical effector position to differ from the logical effector position. Also, because the stepper routines do step-counting to track their positions internally, there would need to be extra logic to allow the positions to update without doing the actual motor stepping.

This is one reason why Marlin has so-far avoided allowing any direct "babystepping" of ABC steppers on a Delta. The XYZ position would need to be recalculated using the forward kinematic method described by Johann Rocholl in his PDF on the subject, a relatively complex calculation.

closed 2016-04-23 by jbrazio