MarlinFirmware/Marlin #2680: Delta skips 0.1mm steps

boulway, 2015-10-07:
My Setup is: Rostock Mini Pro with Sanguinololu / Atmega1284P
Nema 17 running with 1/16 step on A4988 Pololus

While testing the Z calibration of my Delta printer I recognized that sending G91 G1 Z-0.1 is moving the effector just every second time.
I tested it as well with -0.2 and -0.5 and compared the results with M114.

On every value in this small size it looke like the firmware has problems to calculate the motor instructions.
The value of the coordinates are calculated correctly but the motors does not move every time a instruction is given.

I checked the mechanical system and increased the driver output and reduced the motor speed, but it does not look like an hardware issue.

Maybe the Firmware does not send proper instructions to the stepper every time but counts the values for the position correctly.

Best Regards,
Constantin

clefranc, 2015-10-07:
Almost same config here: Rumba / Atmega2560 with 1/16 for XYZ on DRV8825 Pololus.

Can't replicate this as 0.1 moves every time (even lower).

boulway, 2015-10-07:
I could as well test it with Mattercontroller Z-0.1mm Button or Pronterface Z-0.1mm Button. 
Maybe it could be something with float values that work more precise on Atmega2560. I encountered the same behavior on Atmega644 and Atmega1284.

ErwinRieger, 2015-10-07:
Maybe the effect of 'dropsegments':

```
Configuration_adv.h:370:
const unsigned int dropsegments=5; //everything with less than this number of steps will be ignored as move and joined
```

See also http://forums.reprap.org/read.php?1,153723

boulway, 2015-10-07:
Confirmed!
I changed the parameter to 1 and it moves every time.
I worked 3 years with my Prusa i2 and never got this problem, but Deltas are very different i guess.
Thanks a lot!

closed 2015-10-07 by boulway