MarlinFirmware/Marlin #2682: TWI/Wire feature request to RC2 or later

justmyopinion, 2015-10-07:
Wire library in Arduino has a default speed of 100kHz which set the speed for TWI/I2C services for LCD etc. 
Time has come to speed this up as it is no problem to have TWI running at higher speeds.
Speeds > 400khz are available without any changes, I have tested my own up to 800 kHz and still running.
Therefore it would be nice to have a constant defined in configuration for this like TWI_SPEED =400000.
This could then be used to change wire speeds on the run. Wire lib has a service "setClock( TWI_SPEED) that may be called somewhere or it could be changed in register macro TWBR=TWI_SPEED..

thinkyhead, 2016-02-10:
I'm surprised no one has commented on this. Has this been confirmed to be compatible with various configurations including deltabots and graphical displays? And is it in good shape to submit a PR with this concept?

Bio-Printer, 2016-02-10:
Would a higher wire speed allow the I^2C LCD panels to be updated quicker?   Or more specifically, can the processor spend less time in the LCD Update Code?   If so...   This should be carefully explored.

thinkyhead, 2016-03-16:
I'd like to hear @AnHardt's comments on this. Specifically whether this would mean more time taken away from the stepper ISR.

AnHardt, 2016-03-16:
I have no clue and no hardware to test.
Later he made https://github.com/MarlinFirmware/Marlin/pull/2719
And finally we ended with https://github.com/MarlinFirmware/MarlinDev/pull/260#issuecomment-152870027 ++
Maybe we can reactivate him now.

thinkyhead, 2016-03-18:
It may comfort @justmyopinion to know that we've had some positive personnel changes, and we're taking a more proactive approach to issues and feature requests going forward.

closed 2016-04-28 by jbrazio