MarlinFirmware/Marlin #2683: Bug in RC1/RC2 updating status leds in Viki/Panolulu TWI panels

justmyopinion, 2015-10-07:
Status leds are only updated with changing conditions and not if menues are visited.
Error in lcd_implementation_update_indicators() causing update only at changes in status but menu updates clear status information for some reason.

AnHardt, 2015-10-09:
As far i can see 'lcd_implementation_update_indicators()' is called with 'LCD_UPDATE_INTERVAL' regardless if in the menu or not.

The relevant variables 'ledsprev' and 'leds' are both defined inside  'lcd_implementation_update_indicators()' so the menu has no chance to alter these.

Ahhh - the port is written to HIGH during 'lcd_implementation_init()'. That is called to reinit a crashed display every time the menu is entered.

In
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd_implementation_hitachi_HD44780.h#L365
you could try to replace `lcd.setBacklight(HIGH);` with `lcd_implementation_update_indicators()`
Likely you need a forward declaration of `lcd_implementation_update_indicators()`

During boot the LEDs may show wrong states for a second.

justmyopinion, 2015-10-09:
That is correct LCD is initialized when scrolling menus interfering with led updates.
I just removed ledsprev as it only is checking for some status changes which is not menu changes) as it is only reducing update rate, but if updated with a fair rate problem is solved

AnHardt, 2015-10-09:
Removing ` ledsprev` means calling `lcd.setBacklight(leds)` unnecessarily often. You wanted to reduce the interrupts caused by that?

justmyopinion, 2015-10-09:
:-D
That was just an easy fix when testing...

AnHardt, 2015-10-09:
Is #2683 what you have done, or did you need a forward declaration?

AnHardt, 2015-10-09:
https://github.com/AnHardt/Marlin/archive/fix%232683_status_LEDs_reinit.zip

justmyopinion, 2015-10-10:
This one needs a bit more work...I will take a look into it..

AnHardt, 2015-10-11:
Does it not compile or not work.

justmyopinion, 2015-10-11:
It compiles fine with a forward declaration and sort of also is working as intended, but it has some blimps showing when display is reset on some menu (?) changes, maybe a softer reset (clear?) could do it.

AnHardt, 2015-10-12:
If the pins are cleared somewhere in `lcd.begin()` we could modify `lcd_implementation_update_indicators()` to `lcd_implementation_update_indicators(init = true)` and set `ledsprev` to 0 when init == true. Then the pins are always written correct when init == true.

thinkyhead, 2016-02-24:
@AnHardt @justmyopinion If everyone concurs that #2821 is working well, I don't mind merging that asap. I'd like to get rid of those `#undef` lines if possible, but otherwise it looks great.

AnHardt, 2016-02-28:
@thinkyhead 
Just close this and #2821 and forget about it.
justmyopinion left us. I have no hardware to test with.
It's just about some flickering LEDs
No one else seems to be interested.

closed 2016-03-09 by thinkyhead