MarlinFirmware/Marlin #2692: bed temperature runaway, auto leveling.

pbrgle1, 2015-10-09:
Since i upgraded to auto bed leveling i keep getting thermal runaway errors for my heat bed. The temperature readings freeze and after the THERMAL_PROTECTION_BED_PERIOD the printer disconnects. This issue happens only when i want to print. If i leave the bed set to 100C while not printing there is no problem.

AnHardt, 2015-10-09:
Could your fan blow the bed cold?

Tune
```
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius
#endif
```
in Configuration_adv.h

pbrgle1, 2015-10-09:
no there are no fans blowing on the bed. Before i enabled auto bed leveling and servos everything worked fine. No other changes were made

AnHardt, 2015-10-09:
Anyway the lines above are the key. Tune or switch off.

Possible sources are:
loose connection/ broken cable at heater or thermistor
noise on the sensor lines.

Please observe what happens with the temperature directly before the error. (graph at the host program)

pbrgle1, 2015-10-09:
http://postimg.org/image/bsagnv2wd/ (GRAPH)

the sensor connection is fine. the reading freezes when i start printing. overload because of bed leveling (G29) maybe?

AnHardt, 2015-10-09:
It's difficult to judge in that scale, but as far i can see Marlin is doing what it's expected to do.
The beds temperature is dropping more than 2°C for longer then 20 seconds. Then for safety reasons your printer is shut down.
You can make this knife blunt by increasing the temperature difference and or increasing the time.

pbrgle1, 2015-10-09:
Yes i see what you are saying but why does this not happen when im preheating the bed? I also set it to 40s.

AnHardt, 2015-10-11:
The difference between preheating and printing is moving. 
A broken wire?
Cooling by moving?

ghuser2, 2015-10-12:
When preheating, the thermal runaway protection is not activated yet, until the temp stabilized.

AnHardt, 2015-10-12:
@ghuser2 
Good finding.

If the temperature is still to unstable playing with
```
// Actual temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.
```

and or tuning the PID values can improve the situation.

pbrgle1, 2015-10-13:
In my case the bed is not moving douring printing (Only in Z axis). I had to disable the protection for my bed because no setting worked.

And abut preheating. Even when the temps stabilised and stayed there for some time, i had no problems with thermal runaway. Only when starting a print.

closed 2015-10-13 by AnHardt