MarlinFirmware/Marlin #2693: Feature request for optimizing marlin ISR code.

justmyopinion, 2015-10-09:
During time a lot of people has added functionality by adding code to interrupt routines which is not always a good idea as timing here is critical.
I have one more request to consider here to take some load out of these routines as I think it is one way of getting Marlin back on track as the reliable sw it actually is.
Button wheel routines are called in Timer0 (millis) routines (not slowButtons) which I really think is overkill as no buttons needs an update rate of 1000 times/s (1ms). These updates ( lcd_buttons_update()) could easily be transfered to surrounding idle routines as lcd_update with a much lower update rate freeing valuable processor cycles to other critcal processing jobs.
Actually, button update calls already exist in lcd_update so they can just be removed from ISR routine without any problem at all as turnaround in idle mode has a max rate of app 15000/s (60us) which is even larger than interupt routine of 1ms (1000/s) :-D

Wackerbarth, 2015-10-09:
@justmyopinion -- I heartedly endorse your concern. I think that it is likely the case that even the checking of limit switches can be demoted from the primary ISR to a lower priority. In general, the mechanical system can tolerate a few steps after contacting the switch. And the only place where the exact contact point is important is when attempting to align (home) the actuators. In that case, they can be moved "in slow motion" so that the sampling rate is higher than the movement rate.

At some point, I would also hope that we could resolve the conflict between the stepping interrupt and the processing of serial i/o data. But, recent changes seem to have moved in the wrong direction in that regard.

thinkyhead, 2016-02-24:
We're still awaiting code patches from the deeply concerned, but in the meantime we should figure out what calls we can move into `idle()` which are currently being called from the stepper ISR. There is also a Temperature ISR. And when `ADVANCE` is enabled it starts yet another ISR.

In thinking about the best approach, I'm curious, how often does `loop()` run? Can that period be adjusted? Does the main loop `sleep` between invocations? I guess I should know the basics of `loop` by now…. In any case, it sounds like we can move almost anything at all to `idle()` where the exactness of the timing doesn't matter.

Strictly speaking, this issue should be reposted on the MarlinDev issue queue. It's probably not the kind of change we want to make to the release candidate.

closed 2016-04-23 by jbrazio